PySpice-org/PySpice #2: Problem running examples on Python 2.7

kiryx, 2015-04-14:
Hello,

I have compiled and installed ngspice (version 26) from sources, along with PySpice (Python 2.7.6) from pip (pip2.7).

When I try to execute the following:
```Python
from PySpice import Spice
```
I get the following error:
```
  File "/usr/local/lib/python2.7/dist-packages/PySpice/Spice/Netlist.py", line 284
    class Element(object, metaclass=ElementParameterMetaClass):
                                   ^
SyntaxError: invalid syntax
```

This syntax seems to be a Python3 feature (see https://www.python.org/dev/peps/pep-3115/ and http://www.pythoncentral.io/how-metaclasses-work-technically-in-python-2-and-3/). The PySpice available in pip2 contains the newer (unsupported) syntax.

Does PySpice still support Python 2.7 ?

kiryx, 2015-04-14:
The commit https://github.com/FabriceSalvaire/PySpice/commit/53cde2a606f9c69fe5de305a4a10281ec3365cec answers my question.

closed 2015-04-14 by kiryx

FabriceSalvaire, 2015-04-14:
Yes, I am moving to Python 3 and I decided for this project to don't keep the compatibility with Python 2. I will fix the setup.py