PySpice-org/PySpice #6: Subcircuit parameters not handled correctly

mjfwest, 2015-11-19:
The modified call to subcuircuits with parameters is not implemented as far as I can tell.

The spice code I am trying to arrive at is:
```
X2 7 6 GAIN  K=50m 
.SUBCKT GAIN  1  2 K=20m
*Connections: In Out
*Parameters: K Gain
E1 2 0 1 0 {K}
.ENDS
```
I have the following Python code:
``` python
circuit = Circuit('REGULATOR2')
gain = SubCircuit('GAIN', 1, 2, 'K=20')  
gain.VCVS(1, 2, 0, 1, 0, '{K}')
circuit.subcircuit(gain)
circuit.X(2, 'GAIN', 7, 6, 'K=50m')
print(str(circuit))
```
which gives
```
.title REGULATOR2
.subckt GAIN 1 2 K=20m
E1 1 0 2 0 {K}
.ends
X2 7 6 K=50m GAIN
```
Notice the incorrect order of the name and the parameters.

From http://ngspice.sourceforge.net/docs/ngspice26-manual.pdf

2.4.3 Subcircuit Calls
General form:
```XYYYYYYY N1 <N2 N3 . . . > SUBNAM```
Examples:
```
X1 2 4 17 3 1 MULTI
```
Subcircuits are used in ngspice by specifying pseudo-elements beginning with the letter X, followed by the circuit nodes to be used in expanding the subcircuit. If you use parameters, the subcircuit call will be modified (see 2.8.3).

2.8.3 Subcircuit parameters
General form:
```
.subckt < identn> node node . . . <ident>=<value> <ident>=<value> ... 
```
The syntax of a subcircuit call (invocation) is:
General form:
```X<name> node node . . . <identn> <ident>=<value> <ident>=<value> ...```
Examples:
```X2 7 6 GAIN  K=50m ```

FabriceSalvaire, 2015-11-21:
The example *issues/issue-6.py* should fix your issue.

Notice for Netlist.TwoPortElement, as opposite to Spice, the input nodes are specified before the output nodes. But this is confusing ... and could change.

You could also look at *examples/transformer/transformer.py*.

closed 2015-11-21 by FabriceSalvaire

mjfwest, 2015-11-23:
I don't think the transformer example is complete. The Transformer Module does not seem to be in the repository. 
``` Python
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
PySpice/examples/transformer/transformer.py in <module>()
     17 ####################################################################################################
     18 
---> 19 from Transformer import Transformer
     20 
     21 ####################################################################################################

ImportError: No module named 'Transformer'
```

I'll open another issue #10 for this.

FabriceSalvaire, 2015-11-24:
In this case you have to run the example from its directory, else Python will not found the module.