PySpice-org/PySpice #10: transformer example missing the Transformer module?

mjfwest, 2015-11-23:
I don't think the transformer example is complete. The Transformer Module does not seem to be in the repository. 
``` Python
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
PySpice/examples/transformer/transformer.py in <module>()
     17 ####################################################################################################
     18 
---> 19 from Transformer import Transformer
     20 
     21 ####################################################################################################

ImportError: No module named 'Transformer'
```

FabriceSalvaire, 2015-11-24:
Maybe I should tweak the sys.path ?

mjfwest, 2015-11-24:
OK. I think I figured it out.  My directory is not case sensitive and when I check out the files I end up with either Transformer.py or transformer.py.  Could they be called something different for simplicity?

FabriceSalvaire, 2015-11-24:
Oups it looks like a known Windows issue. Yes I have to rename.

mjfwest, 2015-11-24:
I'm on OSX, and thought I was case sensitive...   Either way it is a fix for Windows.  With a bit more googling, I found it might be worth setting: 
`git config core.ignorecase true`

http://stackoverflow.com/questions/17683458/how-do-i-commit-case-sensitive-only-filename-changes-in-git

FabriceSalvaire, 2015-12-26:
fixed in commit d816fb4a6245ac2d2596693d2be1fa83538538d8

closed 2015-12-26 by FabriceSalvaire