PySpice-org/PySpice #11: BehavorialSource not working.

mjfwest, 2015-11-24:
The Spice code that should be generated is: 
```
.SUBCKT SUM2  1   2   3  K1=1 K2=1
*Connections: In1 In2 Out
*Parameters: K1 GAIN1, K2 GAIN2
B1 3 0 V = {K1}*V(1) + {K2}*V(2)
.ENDS
```
I can find no way of calling of calling BehavorialSource for the voltage or current expression to be included, e.g. 
``` python
summer = SubCircuit('SUM2', 1, 2, K_1=1.0, K_2=1.0)
summer.BehavorialSource(1, 3, summer.gnd, 'V = {K1}*V(1) + {K2}*V(2)')
print(str(summer))
.subckt SUM2 1 2 K_2=1.0 K_1=1.0
B1 3 0 
.ends
```

FabriceSalvaire, 2015-11-24:
You have to look in PySpice/Spice/BasicElement.py or best in the API doc http://fabricesalvaire.github.io/PySpice/api/PySpice/Spice/BasicElement.html

The code should be
```
summer = SubCircuit('Sum', 1, 2, 3, K1=1.0, K2=1.0)
summer.BehavorialSource(1, 3, summer.gnd, voltage_expression='{K1}*V(1) + {K2}*V(2)')
```

The parameter renaming is of course questionable.

FabriceSalvaire, 2015-11-24:
Issue: the doc generated by Sphinx is useless in such cases.

FabriceSalvaire, 2015-11-24:
I uploaded a new version which implement aliasing for spice parameters, thus both (spice and longer one) will work.

closed 2015-12-26 by FabriceSalvaire