PySpice-org/PySpice #12: The element's API documentation is not clear.

FabriceSalvaire, 2015-11-24:
and the Spice parameters are renamed to more expressive and longer ones.

mjfwest, 2015-11-24:
Thanks for the commit and clarification to #11  
I realise now this one was more of an understanding of the documentation problem than a problem with the coding intention, but interpreting the code is difficult because of the fancy Meta Class mechanism you have used.   Would writing doctests be a preferred solution?  I would happily add to them as I go along.

FabriceSalvaire, 2015-11-24:
Yes, Metaclass is powerfull but quite complex for most peoples.

I use a Sphinx extension to generate the documentation from the code, it requires less work but there are some issues. I don't have a solution actually.

Best is to indicate to me when the doc in unclear and improve the module docstring if necessary. API examples could help also.