PySpice-org/PySpice #19: Tube Library Issues

BillScherer, 2016-03-05:
While attempting to use Norman Koren's spice tube library[1], I ran into parsing errors.

When I try to use the full library of tube libs:

```
Tubes = SpiceLibrary('/home/scherbi/src/NK/')
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-4-fdd51c0e1165> in <module>()
      1 #Tubes = SpiceLibrary('/home/scherbi/projects/electronics/library')
----> 2 Tubes = SpiceLibrary('/home/scherbi/src/NK/')

/usr/local/lib/python3.4/dist-packages/PySpice/Spice/Library.py in __init__(self, root_path)
     66             if path.extension.lower() in ('.lib', '.mod'):
     67                 self._logger.debug("Parse {}".format(path))
---> 68                 spice_parser = SpiceParser(path)
     69                 if spice_parser.is_only_subcircuit():
     70                     for subcircuit in spice_parser.subcircuits:

/usr/local/lib/python3.4/dist-packages/PySpice/Spice/Parser.py in __init__(self, path, source)
    556         self._find_sections()
    557 
--> 558     ##############################################
    559 
    560     def _merge_lines(self, raw_lines):

/usr/local/lib/python3.4/dist-packages/PySpice/Spice/Parser.py in _parse(self, lines)
    627                 scope.append(element)
    628 
--> 629         return tokens
    630 
    631     ##############################################

/usr/local/lib/python3.4/dist-packages/PySpice/Spice/Parser.py in __init__(self, line)
    337             number_of_pins = prefix_data.number_of_pins
    338             if number_of_pins:
--> 339                 self._nodes, stop_location = self._line.read_words(stop_location, number_of_pins)
    340         else: # Q or X
    341             if prefix_data.prefix == 'Q':

UnboundLocalError: local variable 'prefix_data' referenced before assignment

```

If I create a new library folder with just one .lib file in it, and just one subcricuit definition in that lib file:
```
Tubes = SpiceLibrary('/home/scherbi/projects/electronics/library')
NameError                                 Traceback (most recent call last)
<ipython-input-5-71e751d108d3> in <module>()
----> 1 Tubes = SpiceLibrary('/home/scherbi/projects/electronics/library')
      2 #Tubes = SpiceLibrary('/home/scherbi/src/NK/')

/usr/local/lib/python3.4/dist-packages/PySpice/Spice/Library.py in __init__(self, root_path)
     66             if path.extension.lower() in ('.lib', '.mod'):
     67                 self._logger.debug("Parse {}".format(path))
---> 68                 spice_parser = SpiceParser(path)
     69                 if spice_parser.is_only_subcircuit():
     70                     for subcircuit in spice_parser.subcircuits:

/usr/local/lib/python3.4/dist-packages/PySpice/Spice/Parser.py in __init__(self, path, source)
    556         self._find_sections()
    557 
--> 558     ##############################################
    559 
    560     def _merge_lines(self, raw_lines):

/usr/local/lib/python3.4/dist-packages/PySpice/Spice/Parser.py in _parse(self, lines)
    627                 scope.append(element)
    628 
--> 629         return tokens
    630 
    631     ##############################################

/usr/local/lib/python3.4/dist-packages/PySpice/Spice/Parser.py in __init__(self, line)
    340         else: # Q or X
    341             if prefix_data.prefix == 'Q':
--> 342                 self._nodes, stop_location = self._line.read_words(stop_location, 3)
    343                 # Fixme: optional node
    344             else: # X

/usr/local/lib/python3.4/dist-packages/PySpice/Spice/Parser.py in read_words(self, start_location, number_of_words)
    470             else:
    471                 if start_location < stop_location:
--> 472                     start_location = stop_location
    473                 else: # we have read a space
    474                     start_location += 1

NameError: Bad element line, looking for word 3/4:
G1 1 3 VALUE={(PWR(V(7),EX)+PWRS(V(7),EX))/KG1}
       ^

These libraries work fine with NGSpice 26.

I have the low pass filter example work inside a Jupyter notebook, and the ability to do circuit simulation in this environment  is fascinating. I am hoping we can get this library to work. I am an experienced Python programmer; I will try to discover the solution(s) to these issue as I have time.

Thanks!

[1] http://www.normankoren.com/Audio/Tubemodspice_article.html

BillScherer, 2016-03-06:
I have narrowed this down, ultimately, to the yet to be implemented _.func_. 

Plus sign (+) based line continuation seems to be a problem as well. Easy enough to work around.

I may be able to work around the unimplemented _.func_ for now; need to test this idea yet.

I'd be willing to work on the .func implementation if that would be helpful.

Thanks.

FabriceSalvaire, 2016-08-11:
Spice parser is certainly far yet complete. Could you package an example showing theses issues?

FabriceSalvaire, 2017-09-22:
Links

http://www.normankoren.com/Audio/Tubemodspice_article.html
http://www.normankoren.com/Audio/Tubemods.zip

FabriceSalvaire, 2017-09-22:
Test parser on lib file

> 
> Traceback (most recent call last):
>   File "./bin/cir2py", line 60, in <module>
>     parser = SpiceParser(path=args.circuit_file)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 699, in __init__
>     self._statements = self._parse(lines)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 770, in _parse
>     element = Element(line)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 421, in __init__
>     self._nodes, stop_location = self._line.read_words(stop_location, number_of_pins)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 600, in read_words
>     ' '*start_location + '^')
> NameError: Bad element line, looking for word 1/4:
> TUBE LIBRARY

> Traceback (most recent call last):
>   File "./bin/cir2py", line 60, in <module>
>     parser = SpiceParser(path=args.circuit_file)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 699, in __init__
>     self._statements = self._parse(lines)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 770, in _parse
>     element = Element(line)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 421, in __init__
>     self._nodes, stop_location = self._line.read_words(stop_location, number_of_pins)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 600, in read_words
>     ' '*start_location + '^')
> NameError: Bad element line, looking for word 3/4:
> E1   7 0  VALUE=

> Traceback (most recent call last):
>   File "./bin/cir2py", line 60, in <module>
>     parser = SpiceParser(path=args.circuit_file)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 699, in __init__
>     self._statements = self._parse(lines)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 770, in _parse
>     element = Element(line)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 421, in __init__
>     self._nodes, stop_location = self._line.read_words(stop_location, number_of_pins)
>   File "/home/fabrice/PySpice/PySpice/Spice/Parser.py", line 600, in read_words
>     ' '*start_location + '^')
> NameError: Bad element line, looking for word 3/4:
> G1   1 3  VALUE={(PWR(V(7),EX)+PWRS(V(7),EX))/KG1*ATAN(V(1,3)/KVB)}

...

FabriceSalvaire, 2017-09-25:
fixed in d39f2b0bfc319fe953e1a88474ebd8924596bfe7

closed 2017-09-25 by FabriceSalvaire