PySpice-org/PySpice #22: Logging.yml issue

nachiket, 2016-08-01:
When running examples, there's a weird error wrt logging.yml missing.. I'm running this in Ubuntu 13.04 with Python 3.3 installed.

```$ python3.3 examples/diode/diode-characteristic-curve.py
/usr/local/lib/python3.3/dist-packages/matplotlib/backends/backend_gtk3agg.py:18: UserWarning: The Gtk3Agg backend is known to not work on Python 3.x with pycairo. Try installing cairocffi.
  "The Gtk3Agg backend is known to not work on Python 3.x with pycairo. "
Traceback (most recent call last):
  File "examples/diode/diode-characteristic-curve.py", line 14, in <module>
    logger = Logging.setup_logging()
  File "/usr/local/lib/python3.3/dist-packages/PySpice/Logging/Logging.py", line 37, in setup_logging
    logging_config_file_name = ConfigInstall.Logging.find(config_file)
  File "/usr/local/lib/python3.3/dist-packages/PySpice/Config/ConfigInstall.py", line 30, in find
    return PathTools.find(config_file, Logging.directories)
  File "/usr/local/lib/python3.3/dist-packages/PySpice/Tools/Path.py", line 54, in find
    raise NameError("File %s not found in directories %s" % (file_name, str(directories)))
NameError: File logging.yml not found in directories ('/usr/local/lib/python3.3/dist-packages/PySpice/Config',)```

nachiket, 2016-08-01:
When I comment out logging, I run into another error

> 
> Traceback (most recent call last):
>   File "examples/diode/diode-characteristic-curve.py", line 25, in <module>
>     libraries_path = os.path.join(os.environ['PySpice_examples_path'], 'libraries')
>   File "/usr/lib/python3.3/os.py", line 676, in __getitem__
>     raise KeyError(key) from None
> KeyError: 'PySpice_examples_path'

ldpgh, 2016-09-30:
Have the same issue with missing `Loggin.yml` in  `/usr/lib/python3.4/site-packages/PySpice/Config`.

Tried the Github-search to find it ... 
- 2x issues found
- 2x source files found using "logging.yml" ... but no reference to "logging.yml" ... not clear why it's not shown

**BUT** it is definitely in the repo ... because it is available at
[https://github.com/FabriceSalvaire/PySpice/tree/master/PySpice/Config](https://github.com/FabriceSalvaire/PySpice/tree/master/PySpice/Config)

My Environment: Win7 + cygwin-64 + (Python3.4 just for PySpice, was difficult to get Numpy installed)
Installation:  `setup-x86_64.exe (cygwin-64)`    or   `python3 -m pip install <module>`

ldpgh, 2016-09-30:
[https://github.com/FabriceSalvaire/PySpice/tree/master/PySpice/Spice/NgSpice/api.h](https://github.com/FabriceSalvaire/PySpice/tree/master/PySpice/Spice/NgSpice/api.h) seems to be the same issue. It's in the repo but missing after the installation.

ldpgh, 2016-09-30:
After the installation ./NgSpice looks like this 
```
$ ls -1  /usr/lib/python3.4/site-packages/PySpice/Spice/NgSpice
 __init__.py
 __pycache__
 Shared.py
$
```

FabriceSalvaire, 2016-09-30:
These files are installed by the following lines in **setup_data.py**:

    data_files=[
        (os.path.join(PySpice_path, 'Config'),
         [os.path.join('PySpice', 'Config', 'logging.yml')]),
        (os.path.join(PySpice_path, 'Spice', 'NgSpice'),
         [os.path.join('PySpice', 'Spice', 'NgSpice', 'api.h')]),
    ],

I don't know why it doesn't work on your platform. Path issue on Windows?

You could also try the[ anaconda distribution](https://www.continuum.io/anaconda-overview) to install Python on Windows.

ldpgh, 2016-09-30:
Keep in mind I'm using cygwin-64 on Win-7.

Installation: `python3 -m pip install PySpice`

AFAI can tell there is not `setup_data.py` in  my `/usr/lib/python3.4/site-packages`.
Checked with `find /usr/lib/python3.4/site-packages -ls | grep -i setup_data`.

BUT more find-calls exposes a surprising location of the missing files ... found it at
- `/usr/lib/python3.4/site-packages/usr/lib/python3.4/site-packages/PySpice/Config/logging.yml`
- ` /usr/lib/python3.4/site-packages/usr/lib/python3.4/site-packages/PySpice/Spice/NgSpice/api.h`

It looks like the path to `./site-package` is inserted twice ... just for the two files marked as missing. No other files in this location.

FabriceSalvaire, 2016-10-01:
According to https://docs.python.org/3.5/distutils/setupscript.html#installing-additional-files, it can happen if the starting / is missing on your platform (which is not a native Posix OS).

Try to execute these Python lines to check it:

    from distutils.sysconfig import get_python_lib
    site_packages_path = get_python_lib()
    print(site_packages_path)

**Note cygwin is quite special and not bug free.**

**This issue has to be checked on a standard install on Windows** (it means using native Windows path).

ldpgh, 2016-10-01:
At this point there is a bumper on the street. No chance to get a Python3 installed within my native Win7 to keep my Py2.7.12 setup clean.

Based on the PySpice Py3-requirement it makes less sense (for my opinion) to check it in my Py2.7.12 native Win7 setup ... except you are saying it's worth an installation check in Py2.7.12.

---
cygwin-64 @ Win7 ... installed Jul/2016 (using most recent packages available)
```
$ python3
Python 3.4.3 (default, May  5 2015, 17:58:45)
[GCC 4.9.2] on cygwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from distutils.sysconfig import get_python_lib
>>> site_packages_path = get_python_lib()
>>> print(site_packages_path)
/usr/lib/python3.4/site-packages
>>>
>>> import os
>>> os.linesep
'\n'
>>> exit()

$ python
Python 2.7.10 (default, Jun  1 2015, 18:05:38)
[GCC 4.9.2] on cygwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from distutils.sysconfig import get_python_lib
>>> site_packages_path = get_python_lib()
>>> print(site_packages_path)
/usr/lib/python2.7/site-packages
>>>
>>> import os
>>> os.linesep
'\n'
>>> exit()
```

FabriceSalvaire, 2016-10-02:
It looks like something goes wrong in **distutils** on your platform since the path is considered as a relative one. It could be related to Cygwin. Unfortunately I don't have Python on Windows to check it.

Anyway you can copy manually  these files to test PySpice further.

ldpgh, 2016-10-02:
After knowing where the files are it is an easy task to copy.

---
Maybe it helps your (or maybe not) ... have scanned my hardware. All Python inst. return absolute paths.

Both Win7Pro installations (Py2.7.12 and a hidden Py3.5.2)
```
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation. Alle Rechte vorbehalten.

>>python
Python 2.7.12 (v2.7.12:d33e0cf91556, Jun 27 2016, 15:19:22) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from distutils.sysconfig import get_python_lib
>>> site_packages_path = get_python_lib()
>>> print(site_packages_path)
C:\Programme\Python\Python2.7.12\Lib\site-packages
>>>
>>> import os
>>> os.linesep
'\r\n'
>>> exit()

>>
>>C:\Programme\Python\Python35\python.exe
Python 3.5.2 (v3.5.2:4def2a2901a5, Jun 25 2016, 22:18:55) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from distutils.sysconfig import get_python_lib
>>> site_packages_path = get_python_lib()
>>> print(site_packages_path)
C:\Programme\Python\Python35\Lib\site-packages
>>>
```
---
Linux PC (Ubuntu 16,04.1, fresh install from iso-image) returns absolute paths as well
```
Py2.7.12 : /usr/lib/python2.7/dist-packages
Py3.5.2 : /usr/lib/python3/dist-packages
```

FabriceSalvaire, 2016-10-02:
Just to clarify, I provide to the installer an absolute path, but it is considered as a relative one in your platform (it is why you get this funny path _/usr/lib/python3.4/site-packages/usr/lib/python3.4/site-packages/..._), which isn't expected.

Could you check, the ispath function return effectively True :

    import os
    os.path.isabs(path)
.

ldpgh, 2016-10-02:
os.path.isabs(path)  ... path is not defined in the this 2-line checker

ldpgh, 2016-10-02:
If you ask for this path ...

```
>>> from distutils.sysconfig import get_python_lib
>>> site_packages_path = get_python_lib()
>>> print(site_packages_path)
/usr/lib/python2.7/site-packages
>>> os.path.isabs(site_packages_path)
True
>>>
```

FabriceSalvaire, 2016-10-02:
Yes for site_packages_path on Cygwin.

endolith, 2017-01-21:
I have the same problems.

>You have to set export PySpice_examples_path=<path to example directory>

Where is this example directory?  It is not installed by pip?  I have to manually download the examples from Github and create this directory myself?  If so, I should say so in the documentation.

> At this point there is a bumper on the street. No chance to get a Python3 installed within my native Win7 to keep my Py2.7.12 setup clean.

Anaconda can install Python 2 and 3 side-by-side.

FabriceSalvaire, 2017-01-22:
> Where is this example directory? It is not installed by pip? I have to manually download the examples from Github and create this directory myself? If so, it should say so in the documentation.

You are right, examples are not installed by pip.

PR #33 (2017-06-29): Use package_data instead of data_files
The previous method using data_files would always install these files in
the system directory, even when doing a user install (setup.py install
--user). get_python_lib() always returns the global python path, so the
manual path calculation would be wrong in this case. Instead make use of
package_data which will install the files in the correct location.

Fixes #22
commit 3921d0c8 (2017-06-29): Use package_data instead of data_files

The previous method using data_files would always install these files in
the system directory, even when doing a user install (setup.py install
--user). get_python_lib() always returns the global python path, so the
manual path calculation would be wrong in this case. Instead make use of
package_data which will install the files in the correct location.

Fixes #22
closed 2017-06-29 by FabriceSalvaire