PySpice-org/PySpice #23: PySpice on windows

zhaofeng-shu33, 2016-11-07:
I am a windows user and I try to install PySpice package to my local win-amd64 machine.
There is few instructions about how to install PySpice on the Documentation Page.
(1)  PyYAML is required but not listed (easy to fix)
(2) Though CFFI is optional(only required if ngspice shared library method is invoked), but I can not compile successfully the source code without installation of CFFI. And Installing of CFFI on windows is not an easy task, as it requires Microsoft Visual C++ compilier installed on local machine. (I guess this is some unnecessary module import when running the examples,but if I use PySpice coding line by line, maybe this problem can be avoided.)
(3) Ngspice is not installed on windows by default. If the official release of Ngspice, I can get the program which can only runs on "ngspice window". Therefore python can not use subprocess to call it as a server. After some efforts I find an edition of Ngspice with console program from its official website.
Still it does not work for running the examples successfully. The error gives is RawFile.py cannot locate the binary data. Finnally I found that RawFile Object uses b'Binary:\n' to match against the starting point of binary data in stdout. However, on windows, every new line begins with \r\n, so it works after I changes the matching string to b'Binary:\r\n'. Maybe you do not test PySpice on windows operating system?

FabriceSalvaire, 2016-11-07:
I never tested PySpice on Windows.

1) PyYAML is listed in setup_data.py. Is it broken ???
2) It is a setup issue. I have to disable the ngspice shared feature when the dependencies are not available.
3) It can be fixed easily.

Do you achieve to make it work ?

mjfwest, 2017-03-13:
Points 2 and 3 seem to still be open.    Can you clarify the easy fix for 2 and 3?

zhaofeng-shu33, 2017-03-14:
For Point 2, I think you can check if there is some module depedency importing cffi in basic usage of pyspice. For Point 3, you can treat the searching differently in RawFile.py Line 241, on windows it works if 
it is " binary_line = b'Binary:\r\n' ". And you can give some annoucement in installation help document for windows that ngspice server should be downloaded from ngspice official website, and it should be server edition.

FabriceSalvaire, 2017-03-21:
I don't have time to check PySpice on Windows. Need someone to do the job.

mjfwest, 2017-03-21:
Happy to help. I'm not  a windows user- but I have access to it.   I would just need some pointers to where you think the easy fix should be done.   Basically we want to translate between '\r\n' and '\n' on different platforms?

FabriceSalvaire, 2017-03-21:
@mjfwest 

Basically we have to test that all examples are working on a standard windows installation. The script 
https://github.com/FabriceSalvaire/PySpice/blob/master/tools/generate-rst-examples.sh can be used for that.

Check list:

* doest it install from pypi  ?
* how to install ngspice ?
* fix end of line : **os.linesep can help for \r\n**
* how to deal with ngspice shared (cffi issue) : how to install it ? / how to disable ?

zhaofeng-shu33, 2017-03-22:
(1) As it can be seen from  (1) and (2), if a windows user install PySpice with pip, it may pop up errors showing some modules are required, I can test this on another machine later.
(2) At current time, a windows user should install all dependencies first before he can built PySpice source successfully.
(4) download link "http://ngspice.sourceforge.net/download.html", choose corresponding binary file under"Ngspice experimental". ngspice server works in console mode.

FabriceSalvaire, 2017-03-22:
Maybe cffi requirement is problematic ?

sonium0, 2017-08-28:
I was able to install pyspice on Anaconda3 on windows in combination with 'ng experimental utf8' from sourceforge. After fixing the binary line, I can run examples without python errors, however simulation results are erratic.

FabriceSalvaire, 2017-08-28:
Could you get an output for a basic simulation ? An hex dump for example.

I know some ngspice options are problematic.

Else we could use ngspice shared. 

>  ngspice 64 bit devel contains 64 bit ngspice binaries with GUI, console and the **shared ngspice** dll. Sources are drawn from the scope-inpcom-6 branch at git.

I don't set it as default on Linux since it provides little advantages.

sonium0, 2017-08-29:
Hi Fabrice,
Which basic simulation should I try?
Else, could you explain how to configure pypsice to use the shared ngspice dll?
I see that the manual is talking about 'shared simulation mode' but this seems to be something different.
A working Windows setup is crucial for industry adoption as far as I can tell.

FabriceSalvaire, 2017-09-04:
I come back ...

You can try a basic voltage divider driven by a AC sin wave. Then we should be able to read the output using numpy and plot a sine wave.

To enable shared mode, you have to use`simulator = circuit.simulator(simulator='shared', ...)`

You will have to fix the library path in PySpice/Spice/NgSpice/Shared.py  _load_library. I am not so sure how to fix this on Window, lib path as arg ?

Do you have Visual C++ ? Which is an issue on Windows.

PySpice or more precisely ngspice was not designed for industry.

FabriceSalvaire, 2017-09-04:
I installed Anaconda on my virtual machine and tested PySpice.

**It requires theses fixes:**
- pip dependencies ? 
- tar contains links
- ANSI codes in log
- SET PySpice_example_path=...
- spice_command=
- \r\n

Apparently the output is not as expected : many **\r\n** in the binary data ???

**We have to ask upstream about this release**

FabriceSalvaire, 2017-09-04:
**Cool !!! It works using shared mode.**

**Just need to fix library path.** 

**But plot names are different**, for example 
- v(v-sweep) -> v-sweep 
- Vinput -> vinput

FabriceSalvaire, 2017-09-05:
fixed in 3b45e787579aeb6e966c50fdf37a8b132a8a159e

closed 2017-09-05 by FabriceSalvaire

aiifabbf, 2019-01-08:
Looks like this problem was not completely solved...

> I was able to install pyspice on Anaconda3 on windows in combination with 'ng experimental utf8' from sourceforge. After fixing the binary line, I can run examples without python errors, however simulation results are erratic.
> 
> input_data = np.fromstring(raw_data, count=number_of_columns*self.number_of_points, dtype='f8')
> in RawFile.py seems to work incorrectly or the generated binary format is different on windows.

I found this problem too on Windows, and traced to this line as well.

> I installed Anaconda on my virtual machine and tested PySpice.
>
> **It requires theses fixes:**
>
>     * pip dependencies ?
>
>     * tar contains links
>
>     * ANSI codes in log
>
>     * SET [...]

It is exactly the problem. Ngspice outputs many linesep in raw data. I tested Ngspice 30 and 29, and they were the same.

It seems naively replacing all `\r\n` to `\n` before sending to numpy works. I tried inserting
```python
raw_data = raw_data.replace(b"\r\n", b"\n")
```
before
```python
input_data = np.fromstring(raw_data, count=number_of_columns*self.number_of_points, dtype='f8')
```
and it works on Windows.

Hope to be replied. @FabriceSalvaire