PySpice-org/PySpice #24: FFI interface not used properly in ngspice shared mode

orjan-i-nilsson, 2016-11-08:
examples/ngspice-shared/voltage-divider.py crashes in simulation. 
Ubuntu 14.04, PySpice 0.3.0, cffi.__version__=='1.8.3', ngspice revision 24

Shared.py:360 (called from ngspiceshared-imp)
        self = ffi.from_handle(user_data) <<<<< says something about garbage data

Change(Shared.py:335)
        self_c = ffi.new_handle(self) <<<<< local variable! will be garbage collected when method returns
to
        self.self_c = ffi.new_handle(self)

and
        ngspice_id_c = ffi.new('int *', self._ngspice_id)
to
        self.ngspice_id_c = ffi.new('int *', self._ngspice_id)

FabriceSalvaire, 2016-11-08:
Indeed this code must be checked and with the latest CFFI release.

FabriceSalvaire, 2017-02-22:
Still missing something ???

> Shared.py:360 (called from ngspiceshared-imp)
> self = ffi.from_handle(user_data) <<<<< says something about garbage data

commit 2e7bc75d (2017-02-22): fix #24
closed 2017-02-22 by FabriceSalvaire