PySpice-org/PySpice #34: Missing introduction.txt from repository?

mjfwest, 2017-08-14:
# Missing introduction.txt from repository. 
From an up-to-date clone of the repo I get the following.
```
python setup.py build
Traceback (most recent call last):
  File "setup.py", line 38, in <module>
    exec(compile(open('setup_data.py').read(), 'setup_data.py', 'exec'))
  File "setup_data.py", line 68, in <module>
    long_description = read_readme('README.txt')
  File "setup_data.py", line 62, in read_readme
    text = merge_include(lines, doc_path)
  File "setup_data.py", line 38, in merge_include
    with open(os.path.join(doc_path, include_file_name)) as f:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\BlahBlaBla\\doc\\sphinx\\source\\introduction.txt'
```
creating the missing text file allows compilation.  I think it probably just needs checking in?

FabriceSalvaire, 2017-08-20:
README is broken.

FabriceSalvaire, 2017-09-06:
fixed in ad255bc852d8716459eca7ff7fbf2dc284bba07d

closed 2017-09-06 by FabriceSalvaire