PySpice-org/PySpice #38: Default Library Path on macOS not correct

ryanbahneman, 2017-09-14:
Trying to run the examples fails with this error message:

```
OSError: cannot load library libngspice.so: dlopen(libngspice.so, 2): image not found.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libngspice.so'
```

I installed ngspice with `brew install libngspice`, and pyspice with pip.
I'm running macOS 10.12.

Looking through the contents of libngspice, I saw that the library installed was 'libngspice.dylib'. Updating the library path to point at that file fixed the issue.

```python
PySpice.Spice.NgSpice.Shared.NgSpiceShared.LIBRARY_PATH = 'libngspice{}.dylib'
```

FabriceSalvaire, 2017-09-14:
Thanks for the info, your are the first one to provide feedback on OS X. As you know Apple isn't cool with OS X virtualisation. And I don't trust my unofficial vm ...

Where comes from libngspice on brew ? If I am right brew is a package system managed by a community of users. Thus this package is not provided by ngspice maintainers but by somebody ???

If I understood the issue, on OS X shared libraries use the extension dylib instead of so. Then the library finder is able to locate the library without to worry on its path. Thus it's an easy fix. Right ???

**Could you explain how to install Python, pip on OS X** I want to update the documentation for OS X.

**Could you confirm PySpice works fine ?**

mjfwest, 2017-09-14:
Hello, 
I have succeeded previously with "home brew" ngspice and pyspice in OS X. I'll have a look at what I did and see if it still works.  There are options for python on OSX, the built in python being one. I would recommend people on OS X (and almost any other platform) to use miniconda, or anaconda python and try and maintain environments that way. I will try a completely clean install soon.

ryanbahneman, 2017-09-14:
Brew is a package manager maintained by a community of users, so the formula may or may not be maintained by the ngspice maintainers. However, this formula pulls the libraries directly from the official project page.

Yes, the library is on the path. I just had to change the file extension. Should be an easy fix.

As @mjfwest commented Miniconda/Anaconda python are great options, but I use brew myself. `brew install python3`.

Yeah, let me know when you have a patch

mjfwest, 2017-09-14:
As @ryanbahneman points out, there are at least three common way of installing or using Python on OS X, the 3rd common one being brewed Python. The bigger point is perhaps that any instructions or install scripts should not make assumptions, or should at least cover the most common situations.

FabriceSalvaire, 2017-09-14:
fixed in 70610e06674fac3cdb04de62099b89fc3d7c1cf5

FabriceSalvaire, 2017-09-14:
will update doc soon

closed 2017-09-14 by FabriceSalvaire

cjmcclellan, 2017-11-11:
I am having a similar issue.  I have the latest install (PySpice 1.1.4) on OSX but still get
`OSError: cannot load library libngspice.dylib: dlopen(libngspice.dylib, 2): image not found.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libngspice.dylib'`

I downloaded ngspice with:
`brew install libngspice`
which installed libngspice.dylib in the proper directory.  I've also tried adding the parent directory of libngspice.dylib to the PATH and PYTHONPATH variables, but still now luck.  Any help would be grateful.

FabriceSalvaire, 2017-11-12:
Unfortunately I cannot test on osx, could be this issue was not fixed ???

PATH and PYTHONPATH are not relevant

cjmcclellan, 2017-11-12:
I don't know about other users, but I never got PySpice working on osx.  I found the libngspice.dylib but python cannot find it.  I tried running my own script with
 `ctypes.find_library(libngspice.dylib)` 
but libngspice.dylib cannot be found.  How exactly does pyspice find the libraries?  Is there another path I should add (maybe LD_LIBRARY_PATH)?  Just trying to figure out how to force pyspice to find libngspice.dylib.

tvrusso, 2017-11-13:
OS X doesn't use LD_LIBRARY_PATH.  There is a variable called DYLD_LIBRARY_PATH that can potentially be used instead.  Unfortunately, ever since OS X El Capitan, they added a "security" layer that refuses to propagate DYLD_LIBRARY_PATH down into sub processes, so it is entirely possible it won't help.

HGrabas, 2017-11-13:
I confirm that I am able to runn PySpice on OS X High Sierra. I had to install Python 3.
I install it using brew:
```
brew install python3
brew link python3
brew prune && brew postinstall python3
brew install libngspice
```

An install using pip3 the libraries:

```
pip3 install PySpice
pip3 install numpy
pip3 install scipy
pip3 install matplotlib
pip3 install PyYAML
pip3 install cffi
```
Hopefully that can help you too.