PySpice-org/PySpice #39: SpiceParser problem

ignrub, 2017-09-15:
I have from Kicad a simple voltage divider, Divisor.cir: https://ctrlv.it/id/45314/1698504025
I want to read it and simulate in Pyspice using the following code https://ctrlv.it/python/45312/2388326092
I get the following error:
> 2017-09-15 08:17:07,388 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Warning: singular matrix:  check nodes 1 and 1
> 2017-09-15 08:17:07,389 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Note: Starting dynamic gmin stepping

Could you tell me what i am doing wrong?

FabriceSalvaire, 2017-09-15:
I guess ngspice cannot simulate your circuit because the netlist is wrong.

Print the netlist to check it is right. PySpice don't check the netlist, it just make it and pass it to ngspice.

ignrub, 2017-09-15:
This is the netlist:
2017-09-15 13:00:59,898 - PySpice.Spice.Simulation.NgSpiceSharedCircuitSimulator._run - DEBUG - desk
.title None
.subckt Fonte input output 
Vin input output 10V
.ends Fonte
X1 2 1 Fonte
R1 3 1 1K
R2 2 3 2K
.options TEMP = 25°C
.options TNOM = 25°C
.ic 
.tran 100us 3ms 0s
.end

FabriceSalvaire, 2017-09-15:
Your code works on my environment ! But I don't get theses errors which are in fact warnings ???

You should use 0 for ground and an Operational Point analysis but ...

Notice you can always run a netlist generated by PySpice with the ngspice executable. If it fails then the netlist is wrong or ngspice fails to simulate it.

What is your platform ?

ignrub, 2017-09-15:
I am using Ubuntu 17.04, python 3.6.2 and ngspice 26, here the full output:

$  python Divisor_circuit.py 
```
2017-09-15 14:13:15,188 - PySpice.Spice.Parser.Element.__init__ - DEBUG - 
Element X 1 ['2', '1'] ['Fonte'] {}
2017-09-15 14:13:15,189 - PySpice.Spice.Parser.Element.__init__ - DEBUG - 
Element R 1 ['3', '1'] ['1K'] {}
2017-09-15 14:13:15,190 - PySpice.Spice.Parser.Element.__init__ - DEBUG - 
Element R 2 ['2', '3'] ['2K'] {}
2017-09-15 14:13:15,191 - PySpice.Spice.Parser.Element.build - DEBUG - X 1 ['2', '1'] ['Fonte'] {}
2017-09-15 14:13:15,191 - PySpice.Spice.Parser.Element.build - DEBUG - R 1 ['3', '1'] ['1K'] {}
2017-09-15 14:13:15,192 - PySpice.Spice.Parser.Element.build - DEBUG - R 2 ['2', '3'] ['2K'] {}
.title None
.subckt Fonte input output 
Vin input output 10V
.ends Fonte
X1 2 1 Fonte
R1 3 1 1K
R2 2 3 2K

2017-09-15 14:13:15,196 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.new_instance - INFO - New instance for id 0
2017-09-15 14:13:15,715 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.exec_command - DEBUG - Execute command: set nomoremode
2017-09-15 14:13:15,718 - PySpice.Spice.Simulation.NgSpiceSharedCircuitSimulator._run - DEBUG - desk
.title None
.subckt Fonte input output 
Vin input output 10V
.ends Fonte
X1 2 1 Fonte
R1 3 1 1K
R2 2 3 2K
.options TEMP = 25°C
.options TNOM = 25°C
.ic 
.tran 100us 3ms 0s
.end

2017-09-15 14:13:15,718 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.exec_command - DEBUG - Execute command: destroy all
2017-09-15 14:13:15,770 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.send_stat - DEBUG - ngspice-0 send_stat Prepare Deck
2017-09-15 14:13:15,771 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.send_stat - DEBUG - ngspice-0 send_stat Circuit2
2017-09-15 14:13:15,771 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.send_stat - DEBUG - ngspice-0 send_stat Device Setup
2017-09-15 14:13:15,772 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_init_data - DEBUG - ngspice_id-0 send_init_data
2017-09-15 14:13:15,772 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_init_data - DEBUG -   Vector: v.x1.vin#branch
2017-09-15 14:13:15,773 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_init_data - DEBUG -   Vector: V(3)
2017-09-15 14:13:15,773 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_init_data - DEBUG -   Vector: V(1)
2017-09-15 14:13:15,773 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_init_data - DEBUG -   Vector: V(2)
2017-09-15 14:13:15,774 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_init_data - DEBUG -   Vector: time
2017-09-15 14:13:15,774 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.send_stat - DEBUG - ngspice-0 send_stat op
2017-09-15 14:13:15,774 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Warning: singular matrix:  check nodes 1 and 1
2017-09-15 14:13:15,775 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Note: Starting dynamic gmin stepping
2017-09-15 14:13:15,775 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-03 Note: One successful gmin step
2017-09-15 14:13:15,776 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-04 Note: One successful gmin step
2017-09-15 14:13:15,776 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-05 Note: One successful gmin step
2017-09-15 14:13:15,777 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-06 Note: One successful gmin step
2017-09-15 14:13:15,777 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-07 Note: One successful gmin step
2017-09-15 14:13:15,778 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-08 Note: One successful gmin step
2017-09-15 14:13:15,778 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-09 Note: One successful gmin step
2017-09-15 14:13:15,779 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-10 Note: One successful gmin step
2017-09-15 14:13:15,779 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-11 Note: One successful gmin step
2017-09-15 14:13:15,780 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-12 Note: One successful gmin step
2017-09-15 14:13:15,780 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-12 Note: One successful gmin step
2017-09-15 14:13:15,781 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Warning: singular matrix:  check nodes 1 and 1
2017-09-15 14:13:15,781 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Warning: Dynamic gmin stepping failed
2017-09-15 14:13:15,781 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Note: Starting source stepping
2017-09-15 14:13:15,782 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Supplies reduced to   0.0000% Warning: singular matrix:  check nodes 1 and 1
2017-09-15 14:13:15,782 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-02 Note: One successful gmin step
2017-09-15 14:13:15,783 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Trying gmin =   1.0000E-03 Note: One successful gmin step
[... cut ...]
2017-09-15 14:13:15,794 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.plot - DEBUG - vector[1] V(3) type voltage flags real length 0
2017-09-15 14:13:15,795 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.plot - DEBUG - vector[2] V(1) type voltage flags real length 0
2017-09-15 14:13:15,796 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.plot - DEBUG - vector[3] V(2) type voltage flags real length 0
2017-09-15 14:13:15,796 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.plot - DEBUG - vector[4] time type time flags real length 0

```

FabriceSalvaire, 2017-09-15:
Do you have custom spice init file ?

I am aware some stupid package maintainers compile ngspice with all the experimental features enabled ... Fedora was concerned by this issue

Try to run on console **> ngspice netlist.cir** to be sure. I did both : PySpice and ngspice are all ok for me.

If you cannot figure out how to run your netlist, and if you want to do serious simulation, I recommend to you to either recompile ngspice or check what they did on Ubuntu ...

ignrub, 2017-09-15:
With the Ubuntu version I was having the error:
> OSError: cannot load library libngspice.so: libngspice.so: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libngspice.so'
so I had compiled from source using:
```
./configure --with-ngshared
make
sudo make install
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
```
The funny thing is that the example: "7.13.2. Kicad Netlist Parser Example" is perfectly working!!

FabriceSalvaire, 2017-09-15:
Maybe the Ubuntu package doesn't  provide the shared library. You can check the content of the package or run `locate ngspice | grep so`. **I am interested by this info, because Ubuntu is a large audience distribution.**

I don't understand what happen. Did you try to run ngspice executable ?

I doubt, but try to bless the ground as 0
_singular matrix:  check nodes 1 and 1_ suggests a netlist issue

ignrub, 2017-09-15:
I managed to run all the code, it was the 0 node, I thought that the function:
`circuit = parser.build_circuit(ground=0)`
should create the ground node from any number as the example 7.13.2.

Exactly the Ubuntu package doesn't provide the shared library!  
`locate ngspice | grep so` return nothing but the location were I compiled the ngspice. 

```
$ ngspice Divisor.cir 
******
** ngspice-26 : Circuit level simulation program
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Please get your ngspice manual from http://ngspice.sourceforge.net/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Fri May  6 11:16:28 UTC 2016
******

Circuit: x1  1 0 fonte

ngspice 1 -> 
```
Apparently the ngspice is not given any error, I don't know how to use it yet.

FabriceSalvaire, 2017-09-15:
You have to enter **run** in the interpreter and then **plot v(3)**,  look at ngspice manual.

Well I don't understand why your code works for me and not for you. Is there a compilation option that do magic ??? I think no, I used for ngspice 26

```
  --enable-xspice \
  --disable-debug \
  --enable-cider \
  --with-readline=yes \
  --enable-openmp \
  --with-ngshared
```

**Anyway, I am pretty sure there is an issue** in `build(self, circuit, ground=0)`

```
        for node in self._nodes:
            if str(node) == ground:
                node = 0
            nodes.append(node)
```

It should be int and not str. **Could you try ground='1' instead ?**

FabriceSalvaire, 2017-09-15:
God!

From Ngspice manual p47:

> The ground node must be named “0” (zero). For compatibility reason “gnd” is accepted as ground node, and will internally be treated as a global node and be converted to “0”. **Each circuit has to have a ground node (gnd or 0)!** 

But

```
.title None
.subckt Fonte input output 
Vin input output 10V
.ends Fonte
X1 2 1 Fonte
R1 3 1 1K
R2 2 3 2K
.options TEMP = 25°C
.options TNOM = 25°C
.ic 
.tran 100us 3ms 0s
.end
```

works for me ???

does xspice or cider relax this constrain ???

FabriceSalvaire, 2017-09-17:
This bug was introduced in ce7502669c8ec8fe512d2532db728515531d6e98

fixed in c6dfb0aadae01bd2b4efca75a02352d66d1de80e

closed 2017-09-17 by FabriceSalvaire