PySpice-org/PySpice #46: Enhanced KiCad Integration

FabriceSalvaire, 2017-09-19:
https://answers.launchpad.net/kicad/+question/658090

Wayne Stambaugh answer:
> ng-spice has already been directly integrated into KiCad and will be available in the upcoming stable 5 release. I am not sure PySpice buys us anything we don't already have. with ng-spice We still do not have the schematic editor swigged out to python yet so using python in the schematic editor for simulation would have to be run externally which tends to be less than ideal for users.

ferdymercury, 2020-07-20:
Since a couple of weeks, KiCad's eeschema file format has changed, and a Python API is coming soon

ferdymercury, 2020-07-20:
See also: https://gitlab.com/kicad/code/kicad/-/issues/2333

FabriceSalvaire, 2021-05-14:
PySpice can now read a .kicad_sch file and compute the netlist