PySpice-org/PySpice #49: Initial Conditions on Capacitors and Inductors attributes are not working

IgnacioRubioScola, 2017-09-20:
The initial condition from 'ic' parameter is not working, but initializing the voltage node works well (commented). The problem is with inductors, how I can initialize the simulation with current initial conditions?

Here the code for a RC circuit:

```
import matplotlib.pyplot as plt
plt.close('all') #fecha as janelas com gráficos

#import os

import PySpice.Logging.Logging as Logging
logger = Logging.setup_logging()
import numpy as np
from PySpice.Probe.Plot import plot
from PySpice.Spice.Library import SpiceLibrary
from PySpice.Spice.Netlist import Circuit
from PySpice.Unit import *

circuit = Circuit('RCCircuit')
circuit.V('input', 'node_01', circuit.gnd, 15@u_V)
circuit.R(1, 'node_01', 'node_02', 1000@u_kΩ)
circuit.R(3, 'node_02', 'node_03', 0.5@u_Ω)
circuit.C(1,'node_03',circuit.gnd, 50@u_uF, ic=0@u_V)

#cd_node01 = 15
#cd_node02 =0
#cd_node03 =0
Tf =300@u_s
T = 0.1@u_s

simulator = circuit.simulator(temperature=25, nominal_temperature=25)
#simulator.initial_condition(node_01=cd_node01,node_02=cd_node02,node_03=cd_node03)
analysis = simulator.transient(step_time=T/10, end_time=Tf)

figure = plt.figure(1)
plot(analysis['node_01'])
plot(analysis['node_02'])
plot(analysis['node_03'])

```

FabriceSalvaire, 2017-09-20:
Look at ngspice manual **3.2.13 Capacitor or inductor with initial conditions**

It is an XSpice extension and unfortunately A device are not yet implemented.

FabriceSalvaire, 2017-09-20:
A basic implementation should be easy, similar to X

However a complete implementation requires a good understanding of XSpice.

FabriceSalvaire, 2017-09-21:
Devel branch implements partially A device

FabriceSalvaire, 2017-09-25:
Devel branch implements raw spice pass through

closed 2017-09-25 by FabriceSalvaire

solarjoe, 2019-03-01:
Linux64, Py 3.6, PySpice master, ngspice-subprocess, ngSpice 27

I am seeing the same problem. using `initial_condition` it works fine.

As mentioned above, the devel branch seems to fix this issue, but I can't use it.
I need the changes made in https://github.com/FabriceSalvaire/PySpice/issues/110 and they are only in the master branch.