PySpice-org/PySpice #84: Typo in setup.py dependencies

mikeanthonywild, 2017-11-02:
AFAIK `requires` isn't a valid keyword argument for setuptools. This used to be `install_requires` but was changed in 230fc22. I assume this is a typo?

FabriceSalvaire, 2017-11-05:
Oops I don't remember why ? Read outdated doc ???

https://setuptools.readthedocs.io/en/latest/setuptools.html#declaring-dependencies

https://packaging.python.org/tutorials/distributing-packages/#install-requires

`install_requires=['peppercorn'],`

https://packaging.python.org/tutorials/distributing-packages/#python-requires

`python_requires='>=3',`

> In addition, only versions 9.0.0 and higher of pip recognize the python_requires metadata.

FabriceSalvaire, 2017-11-08:
fixed in dc9287fb56aa5167394e62f7197c64a4c7ccf183

FabriceSalvaire, 2017-11-08:
uploaded new version

look at  #85

closed 2017-11-08 by FabriceSalvaire