PySpice-org/PySpice #88: Solar module model

MattiaDallapiccola, 2017-11-21:
Hi everyone,
I'm trying to model a solar cell/module using PySpice for my thesis but I'm a little bit in trouble because I'm new both to Python and Spice. 
The model I'm using is the one reported in figure, the single diode model. 
I'm trying to sweep the voltage in order to obtain the current circulating and, as a consequence, the I-V curve of the module. 
Here is what I'm doing (without results):

ShockleyDiode = ShockleyDiode(Is=6.58e-10, n=1, degree=25, kelvin=None)
circuit = Circuit('Solar module')

circuit.V('input', circuit.gnd,'3', 5@u_V)
circuit.R('s', '2', '3', 0.356@u_Ω) 
circuit.R('sh', '2', circuit.gnd, 312.55@u_Ω)
circuit.D('1', '1', circuit.gnd, model=ShockleyDiode)
circuit.I('pv', circuit.gnd, '1', dc_value=8.17@u_A)
print(circuit)

simulator = circuit.simulator(temperature=25, nominal_temperature=25)
analysis = simulator.dc(Vinput=slice(0, 50, 0.1))

I can't understand if the simulation is running and how figure out the results. Can anyone help me please? 
Sorry for my inexperience and probably stupid questions

[image]

FabriceSalvaire, 2017-11-21:
`print(circuit)` output the Spice code : you can use it to discuss about your simulation on Spice forum

`ShockleyDiode = ShockleyDiode` is wrong
For model, you have to use something like `circuit.model('bjt', 'npn', bf=80, cjc=pico(5), rb=100)`

Your schema is wrong, you need a load

Advice: **To learn how to do with pyspice, take the closest example and run/modify it**

You can also look at openmodelica,I think you don't need a circuit simulator for this.

MattiaDallapiccola, 2017-11-22:
Thank you for your answer. 
To be more clear I have attached a QUCS scheme of what I want to reproduce in PySpice. Is not possible to measure the current without adding a load? I have found that in ngspice it is possible adding a 0-voltage source and measuring current through it.
[image]

I'm trying to adjust the circuit following your suggestion, now it is like this: 

Vinput 3 0 1
Rs 3 2 0.356
Rsh 2 0 312.55
XD1 0 2 CellDiode
Ipv 2 0 8.17
Vmeas 3 2 0
where cell diode is a diode with the characteristics I wanted and Vmeas is the 0-voltage source for current measure. 
I'm using simulator.save('Vmeas') for saving the current throug the Vmeas component but it seems to be similar to the current from the current generator and not depending from the voltage sweep.
Obviously results are not what I was expecting. Do you think I have no chance with that approach and I need to add the load and measure the current there? simulator.save('component name') is right?

FabriceSalvaire, 2017-11-22:
I misunderstood your first schema. ( I still don't understand what you are doing ... )

You can effectively use a 0-voltage as an ammeter (there is a function to do that, but badly documented), but here you can simply derive the current from the voltage across Rs. Then plot everything to check.

MattiaDallapiccola, 2017-11-22:
Ok, now it's a little bit clearer for me so I can probably explain better. I want to sweep the voltage source Vinput from 0 to the open-circuit voltage of the cell and measure the current in the circuit in order to create the I-V curve of the solar cell. 
This is my code:

circuit.V('input', '1', circuit.gnd, 5)
circuit.R('s', '2', '3', .356) 
circuit.R('sh', '3', circuit.gnd, 312.55)
circuit.D('1',circuit.gnd,'3')
circuit.I('pv','3', circuit.gnd, 8.17)
circuit.V('meas', '1','2', 0 )
simulator = circuit.simulator(temperature=25, nominal_temperature=25)
analysis = simulator.dc(Vinput = slice(0, 50, 0.01))

If i have well understood, I would be able to find the current into analysis.branches. The problem is that is not what I expected. I was simply asking if you have any suggestion or if you see an error in what i'm doing. 
I have created the circuit netlist in KiCad so I think that the problem is not here.
Thank you again

MattiaDallapiccola, 2017-11-23:
I have fixed the problem but I have only last question. When I define the circuit it is possible to create a netlist with inside the model properties instead of having references to external libraries? For example 
...
...
D1      3   0 1n4007
.model 1n4007 D(is=7.02767n rs=0.0341512  n=1.80803 eg=1.05743 xti=5
+ bv=1000 ibv=5e-08 cjo=10p vj=0.7 m=0.5 fc=0.5 tt=1e-07 kf=0 af=1)

FabriceSalvaire, 2017-11-23:
I think you want something like

`circuit.model('bjt', 'npn', bf=80, cjc=pico(5), rb=100)`

MattiaDallapiccola, 2017-11-24:
Exactly! 
In particular I would be able to change the diode properties but I'm still not able to build the circuit.
If I use:
circuit.D('1','3', circuit.gnd, '1n5822' )
circuit.model('1n5822', 'D', is=7.02767, rs=0.0341512, n=1, eg=1.05743, xti=5, bv=1000, ibv=5e-08, cjo=10, vj=0.7, m=0.5, fc=0.5, tt=1e-07, kf=0, af=1)
I cant't run the code because 'is' is recognize as an operator. Moreover, if I set only 'rs' the resulting circuit is:

.title Solar module
Vinput out 0 0
Rs out 3 0.0064558
Rsh 3 0 785.94
D1 3 0 
Ipv 0 3 5.87
.model 1n5822 D (rs=7.02767)
But there is no reference to the model in the Diode component. How can I fix that?

FabriceSalvaire, 2017-11-28:
You found an issue !

MattiaDallapiccola, 2017-11-28:
Fortunately easy to solve!