PySpice-org/PySpice #92: Support removing elements from a circuit

fb39ca4, 2017-12-01:
I'd like to be able to isolate parts of a circuit for measuring impedances, etc. by removing elements, and looking through the API documentation for Netlist I can't see a way to do it.

FabriceSalvaire, 2017-12-02:
**Interesting feature !**

Devel branch has an improved netlist support. Maybe an **enable flag** on element, a kind of "**group**" feature and a **visitor** API would be nice ???

**Actually**, thanks to Python,

**1)** you can monkey patch Netlist with such code :

```
  def remove_element(self, element):
            del self._elements[element.name]
            self._dirty = True
```

**2)** or iterate on` _elements` using `element_iterator` and copy some elements to a new circuit.

**3)** or patch` Netlist.__str__` and Element for enable flag.

FabriceSalvaire, 2018-02-07:
fixed in 6c48fb1

closed 2018-02-07 by FabriceSalvaire