PySpice-org/PySpice #95: Setup Link between PySpice and machine ngspice and git clone of PySpice

GProtoZeroW, 2018-01-08:
I am using python 3.6  ngspice-27 and a fresh pip install clone of PySpice on a Linux mint machine. 

Mr. Salvaire, 
I am sure this is a recurrent issue, but I am having trouble getting PySpice to work. I have installed ngspice-27 on my machine using the steps and config setting you listed on https://pyspice.fabrice-salvaire.fr/installation.html#ngspice-compilation (save having to use sudo install ) and have verified that it is working by calling a simple rc circuit from ngspice terminal access. I then proceeded to clone PySpice to my machine and then run the ```pip install git+https://github.com/FabriceSalvaire/PySpice``` to install PySpice. And there where no issues there and I can call up the library.

What I am having troupe is running the library and the docs in https://pyspice.fabrice-salvaire.fr/faq.html#how-to-set-the-ngspice-library-path and others are not clear to me how to get everything configured. 

here are the errors and the locations of the requested file on my machine if you could advise me as what module and class I need to send this info to in python or what .py file I need to edit in the clone of the PySPice repo and do a rebuild so as I can start using your very promising library it would be much appreciated 
-------------------------------------------------------------------------------------------------------------------------------------

```libraries_path = os.path.join(os.environ['PySpice_examples_path'], 'libraries')
spice_library = SpiceLibrary(libraries_path)
``` 
I have never set eviorment variables in linux before so do I need to be in `PySpice/PySpice/examples` in the cloned folder to set the PWD and also the `'PySpice_examples_path'  or can I do this in a notebook anywhere on my machine 

NameError: name 'libraries_path' is not defined
I have no idea what class call or .py needs to be modified I only know the following about ngspice on my machine

terminal:
```
whereis ngspice
ngspice: /usr/local/bin/ngspice /usr/local/lib/ngspice

whereis libngspice.so
libngspice: /usr/local/lib/libngspice.la /usr/local/lib/libngspice.so
```

FabriceSalvaire, 2018-01-08:
As far I understood your issue, you are looking for https://pyspice.fabrice-salvaire.fr/example-introduction.html#how-to-run-these-examples

I will be fixed in 1.2

GProtoZeroW, 2018-01-10:
Yes I have looked at that page, do the environment vars need to be set from the clone directory before running setup.py 

and beyond the examples in order for PySpice to find ngspice wich directory do i need to set things too. And where do I set those in the library

GProtoZeroW, 2018-01-30:
So I figured this out. If NgSpice and PySpice have all been installed successfully then the following would hold. For reference, I will be referring to the [Ex 8.2.1. Diode Characteristic Curve](https://pyspice.fabrice-salvaire.fr/examples/diode/diode-characteristic-curve.html).

In that example one will see in the first code block:
```
libraries_path = os.path.join(os.environ['PySpice_examples_path'], 'libraries')
spice_library = SpiceLibrary(libraries_path)
```
What this code is asking for is  the location of  https://github.com/FabriceSalvaire/PySpice/tree/master/examples/libraries in PySpice's github directory. 

So in order to get the examples working one needs to copy said folder to a local location. I personally reformat the code as follows

```
libraries_path ='SpiceLib/'
spice_library = SpiceLibrary(libraries_path)
```
where `'SpiceLib/'` is where I keep my spice model files for the cirucit I am working on. Right now the Spice Library parser `SpiceLibrary` supports '.lib', '.mod', '.lib@xyce', '.mod@xyce' files contianing spice model as per the docs https://pyspice.fabrice-salvaire.fr/api/PySpice/Spice/Library.html?highlight=spicelibrary

If you are reworking the Diode example then create a folder partial to Python/Notebook file you are working in. In that folder, you must have the  `1N4148.lib` found in https://github.com/FabriceSalvaire/PySpice/tree/master/examples/libraries/diode/switching. Then just the Diode example should be able to be recreated