PySpice-org/PySpice #104: Can't install. Missing 'Pyterate'

xesscorp, 2018-02-20:
I'm trying to install PySpice from the repo under Win 7, Python 3.6.4. It complains about a missing Pyterate directory:

```
$ pip install git+https://github.com/FabriceSalvaire/PySpice
Collecting git+https://github.com/FabriceSalvaire/PySpice
  Cloning https://github.com/FabriceSalvaire/PySpice to c:\users\devb\appdata\local\temp\pip-hcwa1ore-build
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\DEVB\AppData\Local\Temp\pip-hcwa1ore-build\setup.py", line 51, in <module>
        exec(compile(open('setup_data.py').read(), 'setup_data.py', 'exec'))
      File "setup_data.py", line 29, in <module>
        with open(init_path) as fh:
    FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\DEVB\\AppData\\Local\\Temp\\pip-hcwa1ore-build\\Pyterate\\__init__.py'

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\DEVB\AppData\Local\Temp\pip-hcwa1ore-build\
```

FabriceSalvaire, 2018-02-20:
oops I did a quick copy-past ...

closed 2018-02-20 by FabriceSalvaire

xesscorp, 2018-02-20:
Fixed. Thanks!

commit 30bf16b2 (2018-02-20): typo in setup_data.py, fixed #104