PySpice-org/PySpice #105: ModuleNotFoundError: No module named 'PySpice.Doc'

JoseACarrasco, 2018-03-24:
### Environment (OS, Python version, PySpice version, simulator)
OS X, Python 3.6, PySpice version 1.1.4
### Expected Behaviour
Execution of simulation for a circuit in the examples that contains the line
from PySpice.Doc.ExampleTools import find_libraries
### Actual Behaviour
Python answers that 
ModuleNotFoundError: No module named 'PySpice.Doc'

How can I fix this?

FabriceSalvaire, 2018-03-29:
PySpice version should be 1.2 now, but I did not have time to push it yet ...

Did you install from github or from pip ?

emxys1, 2018-04-03:
Hi Fabrice,
Thanks a lot for your feedback.
I've got 2 MacBooks and I installed PySpice in one with pip and in the other with git. I've just checked and both are version 1.1.4.
Everything sees to install correctly and some simulations work (for instance fft.py) but as soon as I run a file that contains "from PySpice.Doc.ExampleTools" (such as diode-characteristic-curve.py) python gives an error.

I'll try to download version 1.2 from GitHub and install it.

Jose