PySpice-org/PySpice #106: Complex Numbers Not Returned

GProtoZeroW, 2018-04-13:
### Environment (OS, Python version, PySpice version, simulator)
Linux Mint 
Python 3.6.3
PySpice 1.2
NgSpice-27 
### Expected Behaviour
Complex number with phase info returned
### Actual Behaviour
Complex number with phase info are not returned and a warning is generated

### Steps to reproduce the behavior
When I downgrade to PySpice 1.1 no problem but I lose the upgraded syntax and features in 1.2
I have posted a Jupyter Notebook with the results of the RLC example as ran on my computer showing the issue here  https://github.com/GProtoZeroW/IssueNotebooks/blob/master/PySpiceComplexIssueApril2018.ipynb

FabriceSalvaire, 2018-04-16:
Thanks !

I am afraid there are still issues for complex in unit framework ...

closed 2018-04-16 by FabriceSalvaire

GProtoZeroW, 2018-04-25:
Fabrice, 
did you fix this or is it a still to do. Asking cause I just found out from the SiKit-RF folk on how to load my S measurements into there library that will plot Smith charts and do parameter conversion. 
Let me know thanks

FabriceSalvaire, 2018-04-25:
it is fixed in git

commit 903d3ec2 (2018-04-16): fix #106