PySpice-org/PySpice #107: generation of multiple subcircuit of same class

HGrabas, 2018-04-18:
### Environment (OS, Python version, PySpice version, simulator)

### Expected Behaviour
I am writing a child class of subcircuitfactory used to create multiple proportional integrator PI blocks:

That PI block should have programmable parameters `K,KI,AWG,MINV,MAXV` . After reading the documentation I came to the understanding that I should simply add them as params to the __init__() method.

```
class PI(SubCircuitFactory):
  __name__ = 'PI'
  __nodes__ = ('In+', 'In-','Out')
  
  def __init__(self,K,KI,AWG,MINV,MAXV):
    super().__init__()
    self.BehavioralSource('Vprop','prop',self.gnd,voltage_expression ='{}*(V(In+)-V(In-))'.format(K))
    self.BehavioralSource('Vsum','sum',self.gnd,voltage_expression ='V(prop)+V(integ)')
    self.BehavioralSource('Vdiffclip','overfl',self.gnd,voltage_expression ='{}*(V(Out)-V(sum))'.format(AWG))
    self.BehavioralSource('Vsum2','antiwind',self.gnd,voltage_expression ='V(overfl)+V(prop)')
    self.BehavioralSource('VClip','Out',self.gnd,voltage_expression ='V(sum)< {0} ? {0} : V(sum) > {1} ? {1} : V(sum)'.format(MINV,MAXV))
    
    #integrator part - fixme with int model from NGSpice?
    self.VoltageControlledCurrentSource('I_Int','antiwind',self.gnd,self.gnd,'integ',multiplier=KI)
    self.C('i','integ',self.gnd,1)
    self.R('i','integ',self.gnd,1000000) 
```

I would expect that multiple instantiation of that class would create multiple subckt in my SPICE netlist. ( `KG1,KIG1,AWG1,MING1,MAXG1` and `KG2,KIG2,AWG2,MING2,MAXG2` ) are defined elsewhere in the code.

```
circuit.subcircuit(PI(KG1,KIG1,AWG1,MING1,MAXG1))
circuit.X('PI_sensor', 'PI', 'CCDSetPoint', 'L3_CCD', 'HeaterCtrl')

circuit.subcircuit(PI(KG2,KIG2,AWG2,MING2,MAXG2))
circuit.X('PI2_sensor', 'PI', 'PlateSetPoint', 'CryoPlate', 'PlateCtrl')
```

### Actual Behaviour

The actual behavior is that I only get one instance of the subckt PI with only one set of parameters.

```
.subckt PI In+ In- Out
BVprop prop 0 v=10*(V(In+)-V(In-))
BVsum sum 0 v=V(prop)+V(integ)
BVdiffclip overfl 0 v=4*(V(Out)-V(sum))
BVsum2 antiwind 0 v=V(overfl)+V(prop)
BVClip Out 0 v=V(sum)< 0 ? 0 : V(sum) > 20 ? 20 : V(sum)
GI_Int antiwind 0 0 integ m=100
Ci integ 0 1
Ri integ 0 1000000
.ends PI
```

I am not entirely sure that this isn't an issue due to the way I am implementing subcircuit in PySpice. 
But an example of how to do multiple instantiation of the same su circuit with different parameters would be beneficial.

### Steps to reproduce the behaviour

FabriceSalvaire, 2018-04-19:
Ah it's an issue indeed, actually subcircuits are registered with a **name** set by `__name__` from SubCircuitFactory

```
class Netlist:
    def subcircuit(self, subcircuit):
        self._subcircuits[str(subcircuit.name)] = subcircuit

class SubCircuitFactory(SubCircuit):
    __name__ = None
    __nodes__ = None
    def __init__(self, **kwargs):
        super().__init__(self.__name__, *self.__nodes__, **kwargs)
```

Thus you should get the last one ( first one is overwritten ), which is an unexpected behaviour.

**We have to find the right way to provide a different name.**

Is

```
circuit.subcircuit(name, cls)
```

a better option ?

FabriceSalvaire, 2019-03-10:
added doc ed7cc69ca60fa9539950cf0096ab92353230ba3e

**but don't say API is prefect !**

closed 2019-03-10 by FabriceSalvaire