PySpice-org/PySpice #110: RawFile.py and NgSpice/RawFile.py

dooboolacky, 2018-04-29:
### Environment (OS, Python version, PySpice version, simulator)
Linux, python3.5, PySpice-1.2.0, ngspice-subprocess

### Expected Behaviour

### Actual Behaviour
  File "/usr/local/lib/python3.5/dist-packages/PySpice-1.2.0-py3.5.egg/PySpice/Spice/NgSpice/RawFile.py", line 217, in fix_case
    element_translation = {element.lower():element for element in circuit.element_names()}
TypeError: 'odict_keys' object is not callable

 File "/usr/local/lib/python3.5/dist-packages/PySpice-1.2.0-py3.5.egg/PySpice/Spice/NgSpice/RawFile.py", line 222, in fix_case
    node_translation = {node.lower():node for node in circuit.node_names()}
TypeError: 'dict_keys' object is not callable

File "/usr/local/lib/python3.5/dist-packages/PySpice-1.2.0-py3.5.egg/PySpice/Spice/RawFile.py", line 119, in to_waveform
    return WaveForm(self.simplified_name, self.unit(data), abscissa=abscissa)
TypeError: __new__() missing 1 required positional argument: 'shape'

  File "/usr/local/lib/python3.5/dist-packages/PySpice-1.2.0-py3.5.egg/PySpice/Spice/RawFile.py", line 353, in _to_ac_analysis
    branches=self.branches(),
TypeError: __init__() missing 1 required positional argument: 'internal_parameters'

### Steps to fix (some of) the behaviour

use circuit.element_names  not circuit.element_names()
use circuit.node_names() not circuit.node_names()

return WaveForm(self.simplified_name, self.unit(data), data.shape, abscissa=abscissa)

dooboolacky, 2018-04-29:
File "/usr/local/lib/python3.5/dist-packages/PySpice-1.2.0-py3.5.egg/PySpice/Spice/RawFile.py", line 119, in to_waveform
return WaveForm(self.simplified_name, self.unit(data), abscissa=abscissa)
TypeError: new() missing 1 required positional argument: 'shape'

apparently that should actually be

   return WaveForm.from_unit_values(self.simplified_name, self.unit(data), data.shape, abscissa=abscissa)

otherwise some arrays seem to end up having arrays as their units, rather than simple "V" or whatever.

FabriceSalvaire, 2018-04-30:
Could be ngspice-subprocess is broken ...

minecraft2048, 2018-06-09:
Possibly similar bug: https://github.com/FabriceSalvaire/PySpice/issues/120

FabriceSalvaire, 2018-06-10:
fixed in e4c0342

closed 2018-06-10 by FabriceSalvaire