PySpice-org/PySpice #111: UnicodeDecodeError in setup.py

jordan-rothenheimer, 2018-04-30:
### Environment (OS, Python version, PySpice version, simulator)

Arch Linux 4.14.37-1-LTS, Python 3.6.5, latest PySpice from pip, NgSpice 27

### Expected Behaviour

"$ pip install pyspice" installs pyspice.

### Actual Behaviour

```
$ pip install pyspice
Collecting pyspice
  Using cached https://files.pythonhosted.org/packages/7d/42/89c81404c0d3285ed2d2214fb9acde8730096108ebfab89df7c4d64b36d6/PySpice-1.1.4.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-s13lw6om/pyspice/setup.py", line 38, in <module>
        exec(compile(open('setup_data.py').read(), 'setup_data.py', 'exec'))
      File "setup_data.py", line 68, in <module>
        long_description = read_readme('README.txt')
      File "setup_data.py", line 62, in read_readme
        text = merge_include(lines, doc_path)
      File "setup_data.py", line 40, in merge_include
        text += merge_include(f.readlines(), doc_path, included_rst_files)
      File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xce in position 786: ordinal not in range(128)
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-s13lw6om/pyspice/

```
### Steps to reproduce the behaviour

Try installing PySpice.

I also tried sudo'ing the command and installing from git repositories, both directly via "git clone" and with pip's "git+https://....." syntax, and the same error message showed up. I installed PySpice just a few days ago on a different computer, same OS and Python version and everything, so I think whatever it is only came up recently.

FabriceSalvaire, 2018-04-30:
I suspect an issue in your environment.

jordan-rothenheimer, 2018-05-03:
I can get it to install by deleting the ".. include::" lines in README.txt, but then if I try running code it starts yelling at me about "Circuit.Sinusoidal" not being defined or something... anyway yes probably it is something in my environment. I will try to figure it out. Thanks.

slacrherbst, 2018-06-06:
I see the exact same problem. What was your resolution?

slacrherbst, 2018-06-06:
The errors seems to be caused by these lines in doc/shpinx/source/news.txt:

 * Implemented unit prefix cast `U_μV(U_mV(1))` to easily convert values

And we can now use unit helper like :code:`u_mV` or compute the value of :code:`1.2@u_kΩ / 2@u_mA`.

Removing them causes the error to go away.