PySpice-org/PySpice #114: AttributeError: SinusoidalVoltageSource

joseph210385, 2018-05-16:
### Environment (OS, Python version, PySpice version, simulator)
Win7 Python3.6.5 PySpice1.1.4 
### Expected Behaviour

it always return "AttributeError: SinusoidalVoltageSource"
When I trying to input ""source = circuit.SinusoidalVoltageSource('input', 'in', circuit.gnd, amplitude=amplitude, frequency=50)''

### Actual Behaviour

Traceback (most recent call last):
  File "ring-modulator.py", line 41, in <module>
    modulator = circuit.SinusoidalVoltageSource('modulator', 'in', circuit.gnd, amplitude=1@u_V, frequency=1@u_kHz)
  File "D:\Python36\lib\site-packages\PySpice\Spice\Netlist.py", line 719, in __getattr__
    raise AttributeError(attribute_name)
AttributeError: SinusoidalVoltageSource

### Steps to reproduce the behaviour
How can I fix this? Thanks

zigarrre, 2018-05-19:
I can confirm this bug with Arch Linux, Python 3.6.5 and PySpice 1.1.4.

```
$ python examples/filter/low-pass-rc-filter.py 
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/PySpice/Spice/Netlist.py", line 717, in __getattr__
    return self.__getitem__(attribute_name)
  File "/usr/lib/python3.6/site-packages/PySpice/Spice/Netlist.py", line 710, in __getitem__
    raise IndexError(attribute_name)
IndexError: SinusoidalVoltageSource

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/filter/low-pass-rc-filter.py", line 26, in <module>
    circuit.SinusoidalVoltageSource('input', 'in', circuit.gnd, amplitude=1@u_V)
  File "/usr/lib/python3.6/site-packages/PySpice/Spice/Netlist.py", line 719, in __getattr__
    raise AttributeError(attribute_name)
AttributeError: SinusoidalVoltageSource
```

tadr0, 2018-05-21:
I ran into the same issue. With "modulator = circuit.Sinusoidal('modulator', 'in', circuit.gnd, amplitude=1@u_V, frequency=1@u_kHz)" it worked fine.

FabriceSalvaire, 2018-06-07:
Will upgrade package and doc

FabriceSalvaire, 2018-06-07:
done

jimenez44, 2018-09-29:
Environment (OS, Python version, PySpice version, simulator)
Win10 Python2.7.13 PySpice1.2.0

I'm trying to prove the examples in Docs, but when Execute the example low-pass-rc-filter.

  File "low-pass-rc-filter.py", line 28
    circuit.SinusoidalVoltageSource('input', 'in', circuit.gnd, amplitude=1@u_V)
                                                                                                           ^
SyntaxError: invalid syntax

I try putting this at the beginning but the error persists
#!/usr/bin/env
#-*- coding: utf-8 -*-

arashbarghi20, 2019-06-17:
I run following code and there is no output, looks " circuit.SinusoidalVoltageSource" does not work:

`import matplotlib.pyplot as plt

import PySpice.Logging.Logging as Logging
logger = Logging.setup_logging()

from PySpice.Doc.ExampleTools import find_libraries
from PySpice.Probe.Plot import plot
from PySpice.Spice.Library import SpiceLibrary
from PySpice.Spice.Netlist import Circuit
from PySpice.Unit import *

#libraries_path = find_libraries()
#spice_library = SpiceLibrary(libraries_path)

circuit = Circuit('Transistor')

circuit.V('power', 5, circuit.gnd, 15@u_V)
source = circuit.SinusoidalVoltageSource('in', 'in', circuit.gnd, amplitude=.5@u_V, frequency=1@u_kHz)
circuit.C(1, 'in', 2, 10@u_uF)
circuit.R(1, 5, 2, 100@u_kΩ)
circuit.R(2, 2, 0, 20@u_kΩ)
circuit.R('C', 5, 4, 10@u_kΩ)
circuit.BJT(1, 4, 2, 3, model='bjt') # Q is mapped to BJT !
circuit.model('bjt', 'npn', bf=80, cjc=pico(5), rb=100)
circuit.R('E', 3, 0, 2@u_kΩ)
circuit.C(2, 4, 'out', 10@u_uF)
circuit.R('Load', 'out', 0, 1@u_MΩ)

print(circuit)
figure = plt.figure(1, (20, 10))

simulator = circuit.simulator(temperature=25, nominal_temperature=25)
analysis = simulator.transient(step_time=source.period/200, end_time=source.period*2)

axe = plt.subplot(111)
plt.title('')
plt.xlabel('Time [s]')
plt.ylabel('Voltage [V]')
plt.grid()
plot(analysis['in'], axis=axe)
plot(analysis.out, axis=axe)
plt.legend(('input', 'output'), loc=(.05,.1))

plt.tight_layout()
plt.show()
`
[image]