PySpice-org/PySpice #115: Error in Thevenin-Norton example

kurhar, 2018-05-25:
### Environment (OS, Python version, PySpice version, simulator)
Win10 Enterprise 1709 
Python 3.6.4 :: Anaconda, Inc.
PySpice 1.14
ngspice27

### Expected Behaviour
Successful run of example file:
thévenin-norton-theorem.py
(other examples worked; did not test all)

### Actual Behaviour
raise CDefError("duplicate declaration of struct %s" % name)

CDefError: <cdef source string>:7: duplicate declaration of struct ngcomplex

### Steps to reproduce the behaviour
Run example file

FabriceSalvaire, 2018-06-07:
can you confirm on version 1.2 ?

kurhar, 2018-06-08:
I found the error.

Redownloaded ngspice from [this page](http://ngspice.sourceforge.net/shared.html) (64-bit version), so that I have the ngspice.dll.

The folder in pyspice, where ngspice.dll is searched for is `bin_dll`
The folder where ngspice puts ngspice.dll is `bin-dll`

I just changed the ngspice `bin-dll` to `bin_dll` and it worked.

The installation instructions for windows contain a dead link, maybe link the page above for downloading.

FabriceSalvaire, 2018-06-10:
fixed doc in 6a7167

closed 2018-06-10 by FabriceSalvaire

GrizzleyEasy, 2018-10-04:
> I found the error.
>
> Redownloaded ngspice from [this page](http://ngspice.sourceforge.net/shared.html) (64-bit version), so that I have the ngspice.dll.
>
> The folder in pyspice, where ngspice.dll is [...]

I'm having the same problem.
CDefError: <cdef source string>:7: duplicate declaration of struct ngcomplex
Which version of pyspice are you using and which zip of ngspice did  you download ?

GrizzleyEasy, 2018-10-04:
Also I'm using pyspice 1.2.0 and ngspice-28_dll_64.

commit a0f107b5 (2018-06-10): fix #115