PySpice-org/PySpice #117: OSError: cannot load library 'libngspice.so'

tamenol, 2018-06-01:
### Environment (OS, Python version, PySpice version, simulator)
Linux Ubuntu 16.04
Python 3.5
PySpice 1.1.4

### Expected Behaviour

Running 

```
import PySpice.Logging.Logging as Logging
logger = Logging.setup_logging()

from PySpice.Spice.NgSpice.Shared import NgSpiceShared

ngspice = NgSpiceShared.new_instance()

print(ngspice.exec_command('version -f'))
print(ngspice.exec_command('print all'))
print(ngspice.exec_command('devhelp resistor'))
```

without erros.

### Actual Behaviour

> (venv) speklap:thesis_pcb$ python -i run.py 
> 
> 2018-06-01 15:02:01,206 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.new_instance - INFO - New instance for id 0
> Traceback (most recent call last):
>   File "run.py", line 6, in <module>
>     ngspice = NgSpiceShared.new_instance()
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/PySpice/Spice/NgSpice/Shared.py", line 338, in new_instance
>     instance = cls(ngspice_id=ngspice_id, send_data=send_data)
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/PySpice/Spice/NgSpice/Shared.py", line 356, in __init__
>     self._load_library()
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/PySpice/Spice/NgSpice/Shared.py", line 381, in _load_library
>     self._ngspice_shared = ffi.dlopen(library_path)
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/cffi/api.py", line 141, in dlopen
>     lib, function_cache = _make_ffi_library(self, name, flags)
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/cffi/api.py", line 802, in _make_ffi_library
>     backendlib = _load_backend_lib(backend, libname, flags)
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/cffi/api.py", line 797, in _load_backend_lib
>     raise OSError(msg)
> OSError: cannot load library 'libngspice.so': libngspice.so: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libngspice.so'
> 

### Steps to reproduce the behaviour

I've install pyspice using pip in a virtual environment. ngspice I tried installing through apt-get and through tarball. I can open the ngspice 'console' and use it to simulate but can't use it with python.

I can't locate libngspice.so on my system, also my $LD_LIBRARY_PATH shows up empty.

How can I fix this? Thanks

FabriceSalvaire, 2018-06-07:
can you confirm on version 1.2 ?

tamenol, 2018-06-07:
I've reinstalled it using
$ pip install git+https://github.com/FabriceSalvaire/PySpice

> 2018-06-07 18:10:33,939 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.new_instance - INFO - New instance for id 0
> 2018-06-07 18:10:33,967 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._load_library - DEBUG - Load libngspice.so
> Traceback (most recent call last):
>   File "run.py", line 6, in <module>
>     ngspice = NgSpiceShared.new_instance()
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/PySpice/Spice/NgSpice/Shared.py", line 331, in new_instance
>     instance = cls(ngspice_id=ngspice_id, send_data=send_data)
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/PySpice/Spice/NgSpice/Shared.py", line 349, in __init__
>     self._load_library()
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/PySpice/Spice/NgSpice/Shared.py", line 375, in _load_library
>     self._ngspice_shared = ffi.dlopen(library_path)
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/cffi/api.py", line 141, in dlopen
>     lib, function_cache = _make_ffi_library(self, name, flags)
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/cffi/api.py", line 802, in _make_ffi_library
>     backendlib = _load_backend_lib(backend, libname, flags)
>   File "/home/speklap/Projects/InProgress/thesis_pcb/venv/lib/python3.5/site-packages/cffi/api.py", line 797, in _load_backend_lib
>     raise OSError(msg)
> OSError: cannot load library 'libngspice.so': libngspice.so: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libngs
> pice.so'

$LD_LIBRARY_PATH is still empty. Do I need to insert some path to $LD_LIBRARY_PATH?

FabriceSalvaire, 2018-06-07:
what is the path of libngspice.so on your env ?

tamenol, 2018-06-08:
I had installed ngspice using apt-get. I couldn't find the library myself as well. I purged it and tried installing through git with the correct configure option to create the shared library. I haven't got it installed properly yet this way. Something is still broken on my system it seems. Will report back once I got ngspice working.

FabriceSalvaire, 2018-06-10:
You have to check the shared library is provided in the package, some linux distributions don't include it.

dpdutcher, 2018-06-28:
I've come across the same issue.

libngspice.so is in /usr/local/lib

peteut, 2019-10-05:
Hitting the same error, using Debian buster. The filename is `libngspice.so.0` in this case. 

```
$ sudo ldconfig -p | grep libngspice.so
        libngspice.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libngspice.so.0
```

peteut, 2019-10-05:
Using `ctyles.util.find_library` should probably be used here:

```python
from ctypes.util import find_library

ngspice_lib_filename = find_library('ngspice')
```

oe1rsa, 2019-10-21:
sudo apt-get libngspice0-dev

tadeussilver, 2020-05-25:
I have a similar error when trying to simulate this example
[https://pyspice.fabrice-salvaire.fr/examples/filter/low-pass-rc-filter.html](url)
I get the following error
`2020-05-25 00:55:20,166 - PySpice.Spice.NgSpice.Shared.NgSpiceShared.new_instance - INFO - New instance for id 0
2020-05-25 00:55:20,166 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._load_library - WARNING - Set locale LC_NUMERIC to C

---------------------------------------------------------------------------

OSError                                   Traceback (most recent call last)

<ipython-input-8-475d4280ae1b> in <module>()
----> 1 simulator = circuit.simulator(temperature=25, nominal_temperature=25)
      2 analysis = simulator.ac(start_frequency=1@u_Hz, stop_frequency=1@u_MHz, number_of_points=10,  variation='dec')

8 frames

/usr/local/lib/python3.6/dist-packages/cffi/api.py in _load_backend_lib(backend, name, flags)
    825         if first_error is not None:
    826             msg = "%s.  Additionally, %s" % (first_error, msg)
--> 827         raise OSError(msg)
    828     return backend.load_library(path, flags)
    829 

OSError: cannot load library 'libngspice.so': libngspice.so: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libngspice.so'

`
when trying to install libngspice0 library I get another error
`Processing triggers for libc-bin (2.27-3ubuntu1) ...
/sbin/ldconfig.real: /usr/local/lib/python3.6/dist-packages/ideep4py/lib/libmkldnn.so.0 is not a symbolic link

Errors were encountered while processing:
 libngspice0:amd64`
I have already tried many things, almost none of them works, I am not very good at programming, among other things, I really need the module to work in the circuit simulation because I do not have multisim due to the economic resources, therefore if anyone has any solution I am grateful.

my system details are as follows
ubuntu 18.04, python 3.6.2, pyspice 1.4.2

tamenol, 2020-05-25:
Yeah I'm sorry. This is too long ago for me and I don't know anymore what I did.

tadeussilver, 2020-05-25:
help, I already investigated a lot without result, do you know a better alternative than pyscpice?

FabriceSalvaire, 2020-05-26:
@tadeussilver I pushed an updated installation documentation

https://pyspice.fabrice-salvaire.fr/releases/v1.4/installation.html#on-linux

tell me if it help you

tadeussilver, 2020-05-26:
honestly the situation was overcome for the moment, I explain what I was trying to do due to the pandemic I cannot access the computers of the institute and due to my economic situation I cannot access a good range computer, however I try to carry out my university work efficient however complex circuit simulations do not work for me so look for an alternative called google colab.
If you manage to implement it in Google collaborated effectively, I would like you to let me know and I will try this week, however some modules fail when trying.
If you want to know more about Google, I leave the info here.
https://colab.research.google.com/notebooks/welcome.ipynb

group4pgs, 2021-01-25:
Even I am facing the same problem and It would be great to see it rectified soon

sljunkie, 2021-03-22:
@tadeussilver 
I was wondering if I could use PySpice in a Google Colab environment as a educational tool @ uni, as we've all got gmail accounts. So I read its [Importing libraries](https://colab.research.google.com/notebooks/snippets/importing_libraries.ipynb) snipped. I could install the PySpice module with

`!pip install PySpice`

but colab's server (Ubuntu-based?) couldn't find libngspice / libngspice-dev /libngspice0 / libngpice0-dev:

`!apt-get -qq install -y libngspice0-dev
E: Unable to locate package libngspice0-dev`

cyber-g, 2021-03-23:
Please have a look to the following discourse thread : 

Installation example scripts
https://pyspice.discourse.group/t/installation-example-scripts/117

The Dockerfiles demonstrate how to "manually" install the precompiled library distributed on standard deb repo. 
The precompiled libraries are only available for : 
 - Ubuntu Focal (20.04) and newer : https://packages.ubuntu.com/search?keywords=libngspice0-dev&searchon=names&suite=all&section=all
 - Debian Stretch (9) (using the backports) and newer : https://packages.debian.org/search?suite=all&searchon=names&keywords=libngspice0-dev

If you cannot instantiate a system with those OS version, you must install the library by building it from source.

group4pgs, 2021-04-18:
I am using Ubuntu 20.04 and doesn't seem to work in it.
I have seen it working on Arch based linux, Manjaro and it was pretty smooth and easy there

Unkraut3, 2021-11-05:
I had a similar issue. Updating cffi to the newest version (1.5) solved it for me.

Varat7v2, 2022-05-05:
I am using Ubuntu 20.04 and this worked for me:

```
sudo apt update
sudo apt install libngspice0-dev
```

NOTE: I tried even installing libngspice and libngspice0, but nothing worked.

Varat7v2, 2022-05-06:
> @tadeussilver I was wondering if I could use PySpice in a Google Colab environment as a educational tool @ uni. So I read its [Importing libraries](https://colab.research.google.com/notebooks/snippets/importing_libraries.ipynb) snipped. I could install the PySpice module with
> 
> `!pip install PySpice`
> 
> but colab's server (Ubuntu-based?) couldn't find libngspice / libngspice-dev /libngspice0 / libngpice0-dev:
> 
> `!apt-get -qq install -y libngspice0-dev` `E: Unable to locate package libngspice0-dev`
> 
> And that's a shame, really - looks like that package is only available on Ubuntu [from 20.04LTS onwards](https://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=libngspice0&searchon=names). Can anyone using Ubuntu confirm this?

Hi! Did you find any solution for this issue for Google Colab?

cyber-g, 2022-05-15:
You have to install the library manually on Google Colab (the version ubuntu on it is too old). 
Please use the following commands in your Google Colab notebooks : 
``` python
# libngspice0 (from stretch-backports) depends on libc6 which is missing by default
# pre-install dependency
!apt update && apt -q install libc6-dev 
# Download from debian and install library
!wget http://ftp.fr.debian.org/debian/pool/main/n/ngspice/libngspice0_30.2-1~bpo9+1_amd64.deb && apt install -q ./libngspice0_30.2-1~bpo9+1_amd64.deb
!wget http://ftp.fr.debian.org/debian/pool/main/n/ngspice/libngspice0-dev_30.2-1~bpo9+1_amd64.deb && apt install -q ./libngspice0-dev_30.2-1~bpo9+1_amd64.deb
# Install PySpice
!pip3 install PySpice
```

Then it should work. 
Let me know if it does not.

Varat7v2, 2022-06-01:
@cyber-g, Hi! I followed your instructions, and it solved this problem: `Unable to locate package libngspice0-dev`. But I am getting the following error.
```
2022-06-01 21:57:50,787 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - WARNING - spinit was not found
2022-06-01 21:57:50,787 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Note: can't find init file.
2022-06-01 21:57:50,795 - PySpice.Spice.NgSpice.Shared.NgSpiceShared._send_char - ERROR - Warning - approaching max data size: current size = 13392.891 MB, limit = 2214.602 MB
Traceback (most recent call last):
  File "PEDApp_main.py", line 218, in <module>
    circuit, analysis = pyspice.simulate_ckt(netlist_dict)
  File "/content/drive/MyDrive/Colab Notebooks/PROJECT-ECD-main/myPySPICE_master.py", line 318, in simulate_ckt
    simulator = circuit.simulator(temperature=25, nominal_temperature=25)
  File "/usr/local/lib/python3.7/dist-packages/PySpice/Spice/Netlist.py", line 1292, in simulator
    return CircuitSimulator.factory(self, *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/PySpice/Spice/Simulation.py", line 1167, in factory
    return sub_cls(circuit, *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/PySpice/Spice/NgSpice/Simulation.py", line 99, in __init__
    self._ngspice_shared = NgSpiceShared.new_instance()
  File "/usr/local/lib/python3.7/dist-packages/PySpice/Spice/NgSpice/Shared.py", line 449, in new_instance
    instance = cls(ngspice_id=ngspice_id, send_data=send_data, verbose=verbose)
  File "/usr/local/lib/python3.7/dist-packages/PySpice/Spice/NgSpice/Shared.py", line 480, in __init__
    self._init_ngspice(send_data)
  File "/usr/local/lib/python3.7/dist-packages/PySpice/Spice/NgSpice/Shared.py", line 586, in _init_ngspice
    self._get_version()
  File "/usr/local/lib/python3.7/dist-packages/PySpice/Spice/NgSpice/Shared.py", line 867, in _get_version
    output = self.exec_command('version -f')
  File "/usr/local/lib/python3.7/dist-packages/PySpice/Spice/NgSpice/Shared.py", line 851, in exec_command
    raise NgSpiceCommandError("Command '{}' failed".format(command))
PySpice.Spice.NgSpice.Shared.NgSpiceCommandError: Command 'version -f' failed
```

Thank you in advance for your help.

cyber-g, 2022-06-08:
The error message is similar to #40 
It looks like the installation of the library is not installing all the configuration files for `ngspice`. 
To solve that, `ngspice` must be also installed ( with `apt`). 

Please find the update preliminary commands. 

``` python
# libngspice0 (from stretch-backports) depends on libc6 which is missing by default
# pre-install dependency
!apt update && apt -q install libc6-dev ngspice
# Download from debian and install library
!wget http://ftp.fr.debian.org/debian/pool/main/n/ngspice/libngspice0_30.2-1~bpo9+1_amd64.deb && apt install -q ./libngspice0_30.2-1~bpo9+1_amd64.deb
!wget http://ftp.fr.debian.org/debian/pool/main/n/ngspice/libngspice0-dev_30.2-1~bpo9+1_amd64.deb && apt install -q ./libngspice0-dev_30.2-1~bpo9+1_amd64.deb
# Install PySpice
!pip3 install PySpice
```

I used [Simulation using External Sources](https://raw.githubusercontent.com/PySpice-org/PySpice/master/examples/ngspice-shared/external-source.py) as a test example using _shared ngspice_ since I cannot run your code. 
The simulation on Google Colab works without any warning or error for that example.