PySpice-org/PySpice #121: How do I suppress debug logging?

minecraft2048, 2018-06-09:
### Environment (OS, Python version, PySpice version, simulator)

### Expected Behaviour

### Actual Behaviour

### Steps to reproduce the behaviour

FabriceSalvaire, 2018-06-10:
look in PySpice/Config/logging.yml
also you can change some _logger.debug() to info

FabriceSalvaire, 2018-06-10:
According to https://docs.python.org/3/howto/logging.html#custom-levels it could be a bad idea, but I agree we need a simple mechanism to switch on/off spice logging.

Anyway PySpice logging must be cleaned:
- set info level by default
- check the logging level for each line of codes

FabriceSalvaire, 2018-06-10:
partially fixed in 29cb9af

FabriceSalvaire, 2019-03-09:
fix in c68eee9a8cd1d4292f9d369c01506b3858359a69

FabriceSalvaire, 2021-05-14:
Should be outdated now

closed 2021-05-14 by FabriceSalvaire