PySpice-org/PySpice #127: Implement NonLinearCurrentSource

HGrabas, 2018-07-20:
### Environment (OS, Python version, PySpice version, simulator)

### Expected Behaviour

Being able to define in our ngSpice netlists:

> Gxxx n+ n- value={expr}
> Gxxx n1 n2 TABLE {expression}=(x0,y0) (x1,y1) (x2,y2)
> Gxxx n+ n- ( POLY (nd) ) nc1+ nc1- ( nc2+ nc2- ... ) p0 ( p1 ... )

This can be achieved by re-using the code implemented NonLinearVoltageSource (tested to work):

```
class NonLinearCurrentSource(DipoleElement):

    """This class implements a non-linear current sources.

    .. warning:: Partially implemented

    Spice syntax:

    .. code-block:: none

        Gxxx n+ n- value={expr}
        Gxxx n1 n2 TABLE {expression}=(x0,y0) (x1,y1) (x2,y2)
        Gxxx n+ n- ( POLY (nd) ) nc1+ nc1- ( nc2+ nc2- ... ) p0 ( p1 ... )
        Laplace

    Keyword Parameters:

    Attributes:

      :attr:`transconductance`

    """

    __alias__ = 'NonLinearCurrentSource'
    __prefix__ = 'G'

    #transconductance = ExpressionPositionalParameter(position=0, key_parameter=False)
    
    ##############################################

    def __init__(self, name, *args, **kwargs):

        super().__init__(name, *args, **kwargs)

        self.expression = kwargs.get('expression', None)
        self.table = kwargs.get('table', None)

    ##############################################

    def __str__(self):

        spice_element = self.format_node_names()
        # Fixme: expression
        if self.table is not None:
            # TABLE {expression} = (x0, y0) (x1, y1) ...
            table = ['({}, {})'.format(str_spice(x), str_spice(y)) for x, y in self.table]
            spice_element += ' TABLE {%s} = %s' % (self.expression, join_list(table))
        return spice_element
```

### Actual Behaviour

Currently we can only implement a transconductance source.

### Steps to reproduce the behaviour

User-0007, 2022-10-19:
Hello, 

can we implement a polynomial behavior?  Actually, Pyspice doesn't recognise the nonlinear syntax 

Regards,

benedictjones, 2023-01-01:
I have managed to use B sources perfectly fine (Chapter 5.1 of ngspice 31 manual) in my PySpice code.
For example, if we have two nodes 'in' and 'out', we could do something like:

```
i_express = "%s*pwr(v(in)-v(out), 2) + %s*(v(in)-v(out))" % (a,b)  # define current expression
self.B(name, 'in', 'out', i=i_express)   # create B source element
```

which would implement:

```
if V > 0:
  I = aV^2 + bV
else:
  I = -aV^2 + bV
```
I realise this doesn't address the problem with the Gxxxx: non-linear current source, but might be of some help