PySpice-org/PySpice #129: AttributeError: 'NgSpiceShared' object has no attribute '_simulation_type'

ouening, 2018-09-04:
### Environment (OS, Python version, PySpice version, simulator)

### Expected Behaviour

### Actual Behaviour

### Steps to reproduce the behaviour

ouening, 2018-09-04:
AttributeError: 'NgSpiceShared' object has no attribute '_simulation_type'

ouening, 2018-09-04:
Download the latest source code,go to directory "example", run the file in terminal:
$ python run-examples

then the first error happened:

Traceback (most recent call last):
  File "run-examples", line 36, in <module>
    for topic in os.listdir(examples_path):
FileNotFoundError: [Errno 2] No such file or directory: ''

I edit the source code in file "run-examples":
#examples_path = os.path.dirname(__file__)
examples_path = os.path.dirname(os.path.abspath(__file__))

and then another error happened:

Traceback (most recent call last):
  File "operational-amplifier/astable.py", line 40, in <module>
    simulator = circuit.simulator(temperature=25, nominal_temperature=25)
  File "/usr/lib/python3.7/site-packages/PySpice/Spice/Netlist.py", line 1296, in simulator
    return CircuitSimulator.factory(self, *args, **kwargs)
  File "/usr/lib/python3.7/site-packages/PySpice/Spice/Simulation.py", line 708, in factory
    return sub_cls(circuit, *args, **kwargs)
  File "/usr/lib/python3.7/site-packages/PySpice/Spice/NgSpice/Simulation.py", line 99, in __init__
    self._ngspice_shared = NgSpiceShared.new_instance()
  File "/usr/lib/python3.7/site-packages/PySpice/Spice/NgSpice/Shared.py", line 331, in new_instance
    instance = cls(ngspice_id=ngspice_id, send_data=send_data)
  File "/usr/lib/python3.7/site-packages/PySpice/Spice/NgSpice/Shared.py", line 350, in __init__
    self._init_ngspice(send_data)
  File "/usr/lib/python3.7/site-packages/PySpice/Spice/NgSpice/Shared.py", line 427, in _init_ngspice
    self._simulation_type.time: u_s,
AttributeError: 'NgSpiceShared' object has no attribute '_simulation_type'
To continue press Enter

JeroenvO, 2018-10-02:
Does it also log `<date>- PySpice.Spice.NgSpice.Shared.NgSpiceShared - Shared.ERROR - Unsupported Ngspice version 28` ?

GrizzleyEasy, 2018-10-05:
> Does it also log `<date>- PySpice.Spice.NgSpice.Shared.NgSpiceShared - Shared.ERROR - Unsupported Ngspice version 28` ?
> In that case you should either install ngspice 27 or download PySpice newest version from Git.

Also install ngspice 27 as shared library by using:
$ ./configure --with-ngshared --enable-xspice --enable-cider --enable-openmp --disable-debug

FabriceSalvaire, 2019-03-09:
fixed in ee8e2ef461ee379f12136613a8b83400abc5ab5f

closed 2019-03-09 by FabriceSalvaire