PySpice-org/PySpice #132: os.linesep.encode not needed for Xyce on windows

akapelrud, 2018-10-19:
### Environment (OS, Python version, PySpice version, simulator)
Win 10 64 bit
Python 3.6.6 :: Anaconda, Inc.
PySpice:  current HEAD of master branch
Xyce 6.9.0_opensource ('xyce-serial')

### Expected Behaviour
parse `Binary:` line from xyce binary output

### Actual Behaviour
Apparently xyce outputs '\n' on windows, not '\r\n' so it is an error to use os.linesep.encode('ascii')

```--- PySpice/Spice/Xyce/RawFile.py       2018-10-19 11:26:33.020402400 +0200
+++ RawFile.py  2018-10-19 11:26:20.443934300 +0200
@@ -127,7 +127,7 @@

         """ Parse the header """

-        binary_line = b'Binary:' + os.linesep.encode('ascii')
+        binary_line = b'Binary:\n'
         binary_location = output.find(binary_line)
         if binary_location < 0:
             raise NameError('Cannot locate binary data')
```

### Steps to reproduce the behaviour

tvrusso, 2018-10-19:
This is puzzling, as Xyce is printing that "Binary:" with the following C++ line:

    os << "Binary:" << std::endl;

where std::endl is supposed to be the correct line terminator for the operating system wherever it was compiled.  Are you running the Xyce project's Windows build, or are you compiling it yourself under Cygwin?

akapelrud, 2018-10-30:
Yes, that behaviour of `std::endl` is correct. Most editors on Windows these days can handle unix style line endings just fine, so I was wondering why the Xyce team even bothered trying.

I suggest patching this in PySpice to match the current implementation of Xyce, that is use '\n' only. If the Xyce team fixes their code, then PySpice must be fixed later on as well.

tvrusso, 2019-01-08:
It is unlikely that Xyce will be modified to emit windows-specific "\r\n" after the "Binary:" keyword in its rawfiles, given that std::endl will always emit "\n" even on Windows when a file is opened in binary mode.  We didn't "bother trying" (as akapelrud suggests) at all --- binary rawfiles are simply opened in binary mode, and written to using standard C++ I/O primitives, with no regard to what operating system Xyce is running on.  This is probably how it's going to stay.

We also have an ASCII (human readable) rawfile format that is not written in binary mode, and this would have the windows-style line terminators (but no "Binary:" keyword --- it would have a "Values" keyword instead).  This is selected by adding "-a" to the command line when invoking Xyce.

FabriceSalvaire, 2019-03-11:
I installed Xyce 6.10 on my Windows machine and checked v1.3.2 works.

closed 2019-03-11 by FabriceSalvaire