PySpice-org/PySpice #134: PySpice raises Python exceptions when it reads some warnings from ngspice

aiifabbf, 2018-11-01:
### Environment (OS, Python version, PySpice version, simulator)

Ubuntu 16.04, Python 3.6.5, PySpice 1.2.0, ngspice 26 as subprocess

### Expected Behaviour

PySpice should not raise Python exceptions when encountered with warnings in ngspice, because the simulation can still go on in ngspice interactive mode.

### Actual Behaviour

When ngspice works under interactive mode and encounters a warning, it prints the warning but does not stop the simulation. 
```
******
** ngspice-26 : Circuit level simulation program
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Please get your ngspice manual from http://ngspice.sourceforge.net/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Sun Feb  7 10:53:02 UTC 2016
******

Circuit: *sheet name:/opa_sr

Warning: Model issue on line 6160 : .model nch.2 nmos ( lmin=    1.19999999999999995e-06   l ...
  unrecognized parameter (n) - ignored
unrecognized parameter (tlev) - ignored
unrecognized parameter (tlevc) - ignored
unrecognized parameter (sfvtflag) - ignored
Warning: Model issue on line 7568 : .model pch.3 pmos ( lmin=    4.99999999999999977e-07   l ...
  unrecognized parameter (n) - ignored
unrecognized parameter (tlev) - ignored
unrecognized parameter (tlevc) - ignored
unrecognized parameter (sfvtflag) - ignored
Warning: Model issue on line 7895 : .model pch.7 pmos ( lmin=    4.99999999999999977e-07   l ...
  unrecognized parameter (n) - ignored
unrecognized parameter (tlev) - ignored
unrecognized parameter (tlevc) - ignored
unrecognized parameter (sfvtflag) - ignored
ngspice 2 -> ac dec 100 1 10G
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000

No. of Data Rows : 1001
ngspice 2 -> plot
which variable: Vout
ngspice 2 -> 
```

But when PySpice reads the same warning from ngspice, it raises a Python exception and the simulation pauses.
```
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-358-f69c3c6637ef> in <module>()
----> 1 analysis = simulator.ac(start_frequency=1@u_Hz, stop_frequency=10e+9@u_Hz, number_of_points=100, variation="dec")

~/Desktop/lab-ngspice/lib/python3.6/site-packages/PySpice/Spice/Simulation.py in ac(self, *args, **kwargs)
    745     def ac(self, *args, **kwargs):
    746 
--> 747         return self._run('ac', *args, **kwargs)
    748 
    749     ##############################################

~/Desktop/lab-ngspice/lib/python3.6/site-packages/PySpice/Spice/NgSpice/Simulation.py in _run(self, analysis_method, *args, **kwargs)
     74         super()._run(analysis_method, *args, **kwargs)
     75 
---> 76         raw_file = self._spice_server(spice_input=str(self))
     77         self.reset_analysis()
     78         raw_file.simulation = self

~/Desktop/lab-ngspice/lib/python3.6/site-packages/PySpice/Spice/NgSpice/Server.py in __call__(self, spice_input)
    160                             stderr)
    161 
--> 162         return RawFile(stdout, number_of_points)

~/Desktop/lab-ngspice/lib/python3.6/site-packages/PySpice/Spice/NgSpice/RawFile.py in __init__(self, stdout, number_of_points)
    168         self.number_of_points = number_of_points
    169 
--> 170         raw_data = self._read_header(stdout)
    171         self._read_variable_data(raw_data)
    172         # self._to_analysis()

~/Desktop/lab-ngspice/lib/python3.6/site-packages/PySpice/Spice/NgSpice/RawFile.py in _read_header(self, stdout)
    191 
    192         self.circuit_name = self._read_header_field_line(header_line_iterator, 'Circuit')
--> 193         self.temperature, self.nominal_temperature = self._read_temperature_line(header_line_iterator)
    194         self.warnings = [self._read_header_field_line(header_line_iterator, 'Warning')
    195                          for i in range(stdout.count(b'Warning'))]

~/Desktop/lab-ngspice/lib/python3.6/site-packages/PySpice/Spice/RawFile.py in _read_temperature_line(self, header_line_iterator)
    251         # Doing analysis at TEMP = 25.000000 and TNOM = 25.000000
    252 
--> 253         line = self._read_header_line(header_line_iterator, 'Doing analysis at TEMP')
    254         pattern1 = 'TEMP = '
    255         pattern2 = ' and TNOM = '

~/Desktop/lab-ngspice/lib/python3.6/site-packages/PySpice/Spice/RawFile.py in _read_header_line(self, header_line_iterator, head_line)
    221             return line
    222         else:
--> 223             raise NameError("Unexpected line: %s" % (line))
    224 
    225     ##############################################

NameError: Unexpected line: Warning: Model issue on line 6315 : .model nch.4 nmos ( lmin=1.8e-007 lmax=    4.99999999999 ...
```

### Steps to reproduce the behaviour

Include an ngspice model lib that contains parameters undefined in ngspice(but properly defined in some commercial spice-compatible simulators like hspice) and this issue would happen, e.g. tsmc rf18 lib.
```
circuit.raw_spice += ".lib rf018.l tt"
simulator = circuit.simulator(simulator="ngspice-subprocess")
analysis = simulator.ac(start_frequency=1@u_Hz, stop_frequency=10e+9@u_Hz, number_of_points=100, variation="dec")
```

FabriceSalvaire, 2019-03-10:
Thanks to report ! Unfortunately I need the lib to debug this issue but I could not find it using google

aiifabbf, 2019-03-14:
It is a proprietary lib...so I am not sure I can post it here publicly.

I have tested this on another lib, and it is the same problem. This [test.tar.gz](https://github.com/FabriceSalvaire/PySpice/files/2964980/test.tar.gz) contains the lib, the netlist and test code.

The code is as simple as
```python
from PySpice.Spice.Parser import SpiceParser

circuit = SpiceParser("two-stage-amp.cir").build_circuit()
print(circuit) # .lib statement is ignored

circuit.raw_spice += ".lib CMOS_035_Spice_Model.lib tt" # so have to manually add this

simulator = circuit.simulator(simulator="ngspice-subprocess")

analysis = simulator.ac(start_frequency=1, stop_frequency=1e+10, number_of_points=100, variation="dec") # error
```

The error message is
```
Traceback (most recent call last):
  File "plot.py", line 10, in <module>
    analysis = simulator.ac(start_frequency=1, stop_frequency=1e+10, number_of_points=100, variation="dec") # error
  File "/home/benjamin/Desktop/test/env/lib/python3.6/site-packages/PySpice/Spice/Simulation.py", line 747, in ac
    return self._run('ac', *args, **kwargs)
  File "/home/benjamin/Desktop/test/env/lib/python3.6/site-packages/PySpice/Spice/NgSpice/Simulation.py", line 76, in _run
    raw_file = self._spice_server(spice_input=str(self))
  File "/home/benjamin/Desktop/test/env/lib/python3.6/site-packages/PySpice/Spice/NgSpice/Server.py", line 162, in __call__
    return RawFile(stdout, number_of_points)
  File "/home/benjamin/Desktop/test/env/lib/python3.6/site-packages/PySpice/Spice/NgSpice/RawFile.py", line 170, in __init__
    raw_data = self._read_header(stdout)
  File "/home/benjamin/Desktop/test/env/lib/python3.6/site-packages/PySpice/Spice/NgSpice/RawFile.py", line 193, in _read_header
    self.temperature, self.nominal_temperature = self._read_temperature_line(header_line_iterator)
  File "/home/benjamin/Desktop/test/env/lib/python3.6/site-packages/PySpice/Spice/RawFile.py", line 253, in _read_temperature_line
    line = self._read_header_line(header_line_iterator, 'Doing analysis at TEMP')
  File "/home/benjamin/Desktop/test/env/lib/python3.6/site-packages/PySpice/Spice/RawFile.py", line 223, in _read_header_line
    raise NameError("Unexpected line: %s" % (line))
NameError: Unexpected line: Warning: Model issue on line 49 : .model n_33 nmos level=49 acm=3 lmin=3.5e-7 lmax=5.0e-5  ...
```

When I use ngspice from the command line, ngspice will prompt warnings but won't halt simulation
```bash
(env) benjamin@benjamin-Lenovo-IdeaPad-Yoga-13:~/Desktop/test$ ls
b3v3_1check.log           env      test.tar.gz
CMOS_035_Spice_Model.lib  plot.py  two-stage-amp.cir
(env) benjamin@benjamin-Lenovo-IdeaPad-Yoga-13:~/Desktop/test$ ngspice two-stage-amp.cir 
******
** ngspice-26 : Circuit level simulation program
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Please get your ngspice manual from http://ngspice.sourceforge.net/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Sun Feb  7 10:53:02 UTC 2016
******

Circuit: *sheet name:/opa_sr

Warning: Model issue on line 49 : .model n_33 nmos level=49 acm=3 lmin=3.5e-7 lmax=5.0e-5  ...
  unrecognized parameter (acm) - ignored
unrecognized parameter (tref) - ignored
unrecognized parameter (cjgate) - ignored
unrecognized parameter (php) - ignored
Warning: Model issue on line 183 : .model p_33 pmos level=49 acm=3 lmin=3.5e-7 lmax=5.0e-5  ...
  unrecognized parameter (acm) - ignored
unrecognized parameter (tref) - ignored
unrecognized parameter (cjgate) - ignored
unrecognized parameter (php) - ignored
ngspice 2 -> ac dec 100 1 10G
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000

No. of Data Rows : 1001
ngspice 2 -> 
```