PySpice-org/PySpice #147: Installation Problem

zmladen, 2019-02-16:
### Environment (OS, Python version, PySpice version, simulator)

### Expected Behaviour

### Actual Behaviour

### Steps to reproduce the behaviour

zmladen, 2019-02-16:
Hello,

I have followed all the steps for the installation of the PySpice (https://pyspice.fabrice-salvaire.fr/installation.html).
After trying to run the first example (https://pyspice.fabrice-salvaire.fr/examples/advanced-usages/internal-device-parameters.html) I get the following error:

OSError: cannot load library 'C:\Program Files\Spice\bin_dll\ngspice.dll': error 0x7e.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'C:\\Program Files\\Spice\\bin_dll\\ngspice.dll'

According to the documentation to install ngspice under windows all we need to do is to download a ngspice-28_dll_64.zip and unzip it to the “C:\Program Files\Spice64”. I am using windows 10.

For some other examples I get the error:

Traceback (most recent call last):
  File "diode-characteristic-curve.py", line 99, in <module>
    circuit.include(spice_library['1N4148'])
  File "C:\Users\mzec\AppData\Local\Programs\Python\Python36-32\lib\site-packages\PySpice\Spice\Library.py", line 107, in __getitem__
    raise KeyError(name)
KeyError: '1N4148'.

I think that the problem is that the library '1N4148' is missing in the folder. Probably it is possible to give path to the new library folder by "spice_library = SpiceLibrary(some_library_path)"

Thanks for your help.

FabriceSalvaire, 2019-03-10:
should be fixed in v1.3.1

closed 2019-03-10 by FabriceSalvaire