PySpice-org/PySpice #151: Initial steps when charging capacitor are non-exponential

solarjoe, 2019-03-01:
Linux64, Py 3.6, PySpice master, ngspice-subprocess, ngSpice 27

When running the code below the first steps of the calculation seem incorrect:

[image]

This does not depend on the temporal resolution. If the step size is changed the shape does not disappear, it only scales different. With a very fine resolution it seems to disappear, but it is still there.

It is probably related to the numerical method used and is has not fully converged.
Is there a setting in PySpice / ngspice to check the quality of the solution?

```
import PySpice.Spice.Simulation
import matplotlib.pyplot as plt

import PySpice.Logging.Logging as Logging
logger = Logging.setup_logging()

####################################################################################################

from PySpice.Spice.Netlist import Circuit
from PySpice.Unit import *

from PySpice.Probe.Plot import plot

circuit = Circuit('RC circuit, charging capacitor')

capa = 1000@u_uF
resi = 47@u_kΩ

tau = capa * resi

print(tau)

circuit.V('Vin', 'in', circuit.gnd, 10@u_V)
circuit.R(1, 'in', 'out', resi)
circuit.C(2, 'out', circuit.gnd, capa)#, ic=0@u_V)

simulator = circuit.simulator(temperature=25, nominal_temperature=25)

simulator.initial_condition(out=0@u_V )

step_time = 1000@u_ms
analysis = simulator.transient(step_time=step_time, end_time=5*tau)

figure = plt.figure(1, (20, 10))

axe = plt.subplot(121)
axe.grid()
axe.plot(analysis['in'], 'ro')
axe.plot(analysis['out'], 'g.')
axe.set_ylim(0, 11)
axe.set_xlabel('t [s]')
axe.set_ylabel('[V_c]')
axe.legend(('Vin [V]', 'Vout [V]'), loc=(.8,.8))

plt.tight_layout()
plt.show()

```

emrahacar, 2019-03-04:
try 

`simulator = circuit.simulator(temperature=25, nominal_temperature=25, use_initial_condition=True)
`

solarjoe, 2019-03-04:
Thanks for your reply, I tried it, but when I look at the generated string in the debug console the `uic` does not show up:
```
.title Resistor Bridge
Vin in 0 10V
R1 in out 47kOhm
C2 out 0 1000uF
.options TNOM = 25°C
.options TEMP = 25°C
.ic V(out)=0V
.tran 1000ms 235.0s 0s
.end
```

The initial conditions are listed, but the `.tran` line should be 

`.tran 1000ms 235.0s 0s UIC`

Can does the `UIC` show up on your installation?

solarjoe, 2019-03-04:
Ok, at least I found out why the `UIC` is not working, you need to pass it to 

`analysis = simulator.transient(step_time=step_time, end_time=5*tau, use_initial_condition=False)`

and not to

`simulator = circuit.simulator(temperature=25, nominal_temperature=25, use_initial_condition=True`

Something changes, but still not correct.

`use_initial_condition=False`:

[image]

`use_initial_condition=True`:

[image]

???

solarjoe, 2019-03-05:
Ok, I found the problem. I was using matplotlib's `plt.plot` instead of `PySpice.Probe.Plot.plot` and the abscissa did not match.

closed 2019-03-05 by solarjoe