PySpice-org/PySpice #156: Using XSpice to read charge

Naroek56, 2019-04-05:
### Environment (OS, Python version, PySpice version, simulator)
Windows 10, Python 3.6.6, Pyspice 1.3.2, NgSpice 30
### Expected Behaviour
Running simulation and plotting output node.
### Actual Behaviour
Simulation runs indefinitely.
### Steps to reproduce the behaviour
I'm aware that XSpice is only partially implemented, although I do not know whether this should occur or not. I'm using an example from the Ngspice manual. My code is:
`circuit = Circuit("Electrical Circuit")

circuit.parameter("Cl=5n","Cl=5n")
circuit.parameter("Il=100n","Il=100n")
circuit.C(1,"cc",circuit.gnd,"c = Cl")
circuit.I(1,circuit.gnd,1,"Il")
circuit.A("int1","%id(1 cc)","2","time_count")

circuit.model("time_count","int",in_offset=0.0, gain=1.0, out_lower_limit=-1e12, out_upper_limit=1e12, limit_range=1e-9, out_ic=0.0)

simulator = circuit.simulator()
simulator.initial_condition(cc=0)
print(simulator)
analysis = simulator.transient(step_time=100@u_ns, end_time=100@u_us)`

This produces the following simulator (before simulation):
`.title Electrical Circuit
.param Cl=5n
.param Il=100n
C1 cc 0 c = Cl
I1 0 1 Il
Aint1 %id(1 cc) 2 time_count
.model time_count int (gain=1.0 in_offset=0.0 limit_range=1e-09 out_ic=0.0 out_lower_limit=-1000000000000.0 out_upper_limit=1000000000000.0)
.options TEMP = 27°C
.options TNOM = 27°C
.ic V(cc)=0
.end`

If I run this code separately in Ngspice it works, but with PySpice it runs indefinitely. Since the circuit works when I short-circuit 1 and cc I assume that A is not establishing a link properly. 

Support/fix much appreciated!

matthuszagh, 2020-01-14:
Hey, this seems to run ok for me. Are you still experiencing issues? Can you post the full code you use to run the simulation from within PySpice?

Naroek56, 2020-02-11:
Yes, I'm still experiencing issues. I'm using Ngspice 31, Pyspice 1.3.2 and execute the code in JupyterLab. 
This is the full code:
`from PySpice.Probe.Plot import plot
from PySpice.Spice.Netlist import Circuit
from PySpice.Unit import *
from PySpice.Spice.Netlist import SubCircuitFactory

circuit = Circuit("Electrical Circuit")

circuit.parameter("Cl=5n","Cl=5n")
circuit.parameter("Il=100n","Il=100n")
circuit.C(1,"cc",circuit.gnd,"c = Cl")
circuit.I(1,circuit.gnd,"1","Il")
circuit.A("int1","%id(1 cc)","2","time_count")

circuit.model("time_count","int",in_offset=0.0, gain=1.0, out_lower_limit=-1e12, out_upper_limit=1e12, limit_range=1e-9, out_ic=0.0)

simulator = circuit.simulator()
simulator.initial_condition(cc=0)
print(simulator)
analysis = simulator.transient(step_time=100@u_ns, end_time=100@u_us)
plot(analysis["cc"])`

The code works as long as I1 is directly connected to C1. However, when I use the code as written above it breaks the kernel. (message: "The kernel appears to have died")