PySpice-org/PySpice #159: missing libngspice.so

arashbarghi20, 2019-05-29:
> Traceback (most recent call last):
>   File "ngsp_test.py", line 47, in <module>
>     ngspice_shared = MyNgSpiceShared(amplitude=amplitude, frequency=frequency, send_data=False)
>   File "ngsp_test.py", line 21, in __init__
>     super().__init__(**kwargs)
>   File "/home/abbasi/.local/lib/python3.6/site-packages/PySpice/Spice/NgSpice/Shared.py", line 349, in __init__
>     self._load_library()
>   File "/home/abbasi/.local/lib/python3.6/site-packages/PySpice/Spice/NgSpice/Shared.py", line 375, in _load_library
>     self._ngspice_shared = ffi.dlopen(library_path)
>   File "/home/abbasi/.local/lib/python3.6/site-packages/cffi/api.py", line 146, in dlopen
>     lib, function_cache = _make_ffi_library(self, name, flags)
>   File "/home/abbasi/.local/lib/python3.6/site-packages/cffi/api.py", line 828, in _make_ffi_library
>     backendlib = _load_backend_lib(backend, libname, flags)
>   File "/home/abbasi/.local/lib/python3.6/site-packages/cffi/api.py", line 823, in _load_backend_lib
>     raise OSError(msg)
> OSError: cannot load library 'libngspice.so': libngspice.so: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libngspice.so'

I have the mentioned error which I cannot `locate libngspice.so`.

I do appreciate if you can help me to resolve.

oe1rsa, 2019-10-22:
Are you on debian or ubuntu? Try sudo apt install libngspice0-dev

FabriceSalvaire, 2020-05-03:
you could try the new tool ./bin/check-installation

closed 2020-05-03 by FabriceSalvaire