PySpice-org/PySpice #162: transient analysis improvement

thesourcerer8, 2019-08-02:
We are doing standard cell characterization, and have the challenge that we would like to have the simulation run until certain parameters fulfill certain requirements.
At the moment we are doing this by trying to run the simulation for a short period of time, then we check the parameters, then we try again with twice the amount of period time, until the parameters match, or we give up if it is too long.
https://codeberg.org/tok/librecell/src/branch/master/librecell-lib/lclib/characterization/input_capacitance.py#L172
Now the problem is that this way we are doing the same calculations unnecessarily again and again. Is there a better way to do this? Can we define the exit parameters and timeouts upfront and have it run just once and exit immediately when the parameters match? Or can we pause, analyze and continue a simulation?

thesourcerer8, 2019-08-27:
I started researching it now: http://ngspice.sourceforge.net/docs/ngspice-30-manual.pdf#subsection.17.5.79 gives a breakpoint command that might be usable. Perhaps we can set 2 breakpoints, one for the values, the other for a timing-timeout.

FabriceSalvaire, 2020-05-03:
> 17.5.80
>  Stop*: Set a breakpoint
> General Form:
> stop [ after n] [ when value cond value ] ...
> Set a breakpoint. The argument after n means stop after iteration number ‘n’, and the argument
> when value cond value means stop when the first value is in the given relation with the
> second value, the possible relations being

> ...

> Symbol or alias may be used alternatively. All stop commands have to be given in the control
> flow before the run command. The values above may be node names in the running circuit, or
> real values. If more than one condition is given, e.g.
> stop after 4 when v(1) > 4 when v(2) < 2,
> the conjunction of the conditions is implied. If the condition is met, the simulation and control
> flow are interrupted, and ngspice waits for user input.
> In a transient simulation the ‘=’ or eq will only work with vector time in commands like
> stop when time = 200n.
> Internally, a breakpoint will be set at the time requested. Multiple breakpoints may be set. If the
> first stop condition is met, the simulation is interrupted, the commands following run or tran
> (e.g. alter or altermod) are executed, then the simulation may continue at the first resume
> command. The next breakpoint requires another resume to continue automatically. Otherwise
> the simulation stops and ngspice waits for user input.
> If you try to stop at
> stop when V(1) eq 1
> (or similar) during a transient simulation, you probably will miss this point, because it is not
> very likely that at any time step the vector v(1) will have the exact value of 1. Then ngspice
> simply will not stop.