PySpice-org/PySpice #164: basic voltage controlled switch issue

marlowequart, 2019-08-13:
### Environment (OS, Python version, PySpice version, simulator)
on Windows
Python version 3.7
NGSpice version ngspice-30
Pyspice version 1.3.2
Scipy version 1.2.0
matplotlib version 3.0.3

### Expected Behaviour
Here is my script for a basic voltage controlled switch. I would expect that when I turn this switch on, I would get the input voltage on the output of the switch. Instead I am getting a lesser value. There must be some other resistance I am not aware of in one of the models? Where can I find the actual spice model being used by the command "circuit.VoltageControlledSwitch"?? Is there some other way that I can just do a basic voltage controlled switch that connects the input to the output based on a command, without any other interference?

### Actual Behaviour

### Steps to reproduce the behavior

Start Code:
####################################################################################################

import matplotlib.pyplot as plt

####################################################################################################

import PySpice.Logging.Logging as Logging
logger = Logging.setup_logging()

####################################################################################################

from PySpice.Doc.ExampleTools import find_libraries
from PySpice.Probe.Plot import plot
from PySpice.Spice.Library import SpiceLibrary
from PySpice.Spice.Netlist import Circuit
from PySpice.Unit import *

####################################################################################################
# Stuff for model
from PySpice.Spice.NgSpice.Shared import NgSpiceShared

####################################################################################################

libraries_path = find_libraries()
spice_library = SpiceLibrary(libraries_path)

####################################################################################################

circuit = Circuit('Basic Switch')

circuit.PulseVoltageSource('pulse', 'sw_drive', circuit.gnd, 0@u_V, 10@u_V, 1@u_ms, 2@u_ms,)

circuit.V('input', 'input', circuit.gnd, 20@u_V)
circuit.R('load', circuit.gnd, 'sw_node', 5@u_Ohm)
circuit.VoltageControlledSwitch('input','sw_node','sw_drive',circuit.gnd,'sw1',model=None)

#####
# Simulation parameters
#####

# Parameters for basic simulation
simulator = circuit.simulator(temperature=25, nominal_temperature=25)
analysis = simulator.transient(step_time=0.1E-6, end_time=50E-3)

#####
# Plots
#####
NUMBER_PLOTS = '2'

#plots of circuit components
figure = plt.figure(1, (10, 5))
plot1 = plt.subplot(int(NUMBER_PLOTS+'11'))

# Plot of references
plot(analysis.sw_drive, color='r')

plt.grid()
plt.xlabel('t [s]')
plt.ylabel('[V]')
plt.legend(('Switch Drive',''), loc=(.05,.1))

plot2 = plt.subplot(int(NUMBER_PLOTS+'12'))

# Plot of sw drive
plot(analysis.sw_node, color='r')
plot((analysis.sw_node)/circuit['Rload'].resistance,color='b')
plt.grid()
plt.xlabel('t [s]')
plt.ylabel('[V]')
plt.legend(('Switch Output','Load Current'), loc=(.05,.1))

plt.tight_layout()
plt.show()

marlowequart, 2019-09-05:
After doing more testing it seems the VoltageControlledSwitch defaults to an On resistance of 1ohm and an Off resistance of 10ohms. Does anyone know if there is any way to change these resistances in the model? I cannot find anywhere in the source code where this is specified. I tried using my own model like in spice but that does not seem to work either:

circuit.VoltageControlledSwitch(input_plus='sw_drive',input_minus=circuit.gnd,output_minus='sw_node',output_plus='input',name='sw1',model='SW')
circuit.model('SW','SW', Ron=.002@u_Ohm, Roff=1@u_MOhm, Vt=3.0@u_V)

FabriceSalvaire, 2020-05-03:
` SXXXXXXX n+ n- nc+ nc- model <on> <off>` to be completed

FabriceSalvaire, 2020-05-04:
PySpice netlist seems ok

closed 2020-05-04 by FabriceSalvaire

VV-Cephai, 2020-10-01:
Running the code in 
https://github.com/FabriceSalvaire/PySpice/blob/ffb13a6b7c1c494a80064c146a6d18fa42fe0322/issues/issue-164.py

I get the following PySpice netlist:

```
.title Basic Switch
Vpulse sw_drive 0 DC 0V PULSE(0V 10V 0s 0s 0s 1ms 2ms)
Vinput input 0 20V
Rload 0 sw_node 5Ohm
Sinput sw_node sw_drive 0 sw1 switch1
.model switch1 SW (Roff=1MegOhm Ron=0.002Ohm Vt=3.0V)

```
Also the plotted voltage is too low (50mV)
[image]

I think the switch is connected wrong in the netlist. 
`Sinput sw_node sw_drive 0 sw1 switch1`
should be 
`Ssw1 input sw_node sw_drive 0 switch1` (or `Sw1 input sw_node sw_drive 0 switch1`)

`circuit.VoltageControlledSwitch('sw1', 'input', 'sw_node', 'sw_drive', circuit.gnd, model='switch1')`
for 
https://github.com/FabriceSalvaire/PySpice/blob/ffb13a6b7c1c494a80064c146a6d18fa42fe0322/issues/issue-164.py#L42

seems to give the expected result for me.