PySpice-org/PySpice #165: Caching idea

thesourcerer8, 2019-08-14:
I have a usecase where I will most likely need the same simulations to be run again and again very often. To speed this up, I am thinking about a global cache that persistently remembers all Spice runs, and if the same simulation is requested again, it just returns the old result instead of calculating it again. The maximum cache size should be configurable. I am not sure yet, whether that caching will help as much as I hope, but I think in my application it could be helpful: Perhaps the application needs only very small pieces of the results, but storing all the results would be too much storage or even just storing it would slow it down too much. 
Now my question from an architectural point of view is: Where should we implement that caching? In the application that is using PySpice? In PySpice? Or in ngspice or other spice backends? The caching should be robust and have no false positives (that it returns wrong results) and low false negatives (that it cannot recognize that 2 differently looking queries are actually doing exactly the  same calculation).

kdunn926, 2019-12-17:
This might be helpful: https://docs.python.org/3/library/functools.html#functools.lru_cache

It may require some work to ensure the circuit / simulation objects are hashable but it seems achievable.

FabriceSalvaire, 2020-05-03:
There is now a forum to discuss about this important use case

FabriceSalvaire, 2020-05-04:
**We could implement a serialisation of the simulation output to HDF5 and a cache mechanism based on desk checksum (it is used on other projects)**