PySpice-org/PySpice #168: ngspice-subprocess not working

andrewparlane, 2019-09-04:
### Environment (OS, Python version, PySpice version, simulator)

Windows 10 + anaconda (prompt not gui)
Python 3.7.1
Pyspice Version: 1.3.2 (installed via pip)
Simulator: ngspice-subprocess

### Expected Behaviour
Doesn't error.

### Actual Behaviour
Pops up a small ngsipice window (image attached), and nothing happens. When I quit either via the X or the quit button, I get the following errors:

  File "test2.py", line 8, in <module>
    analysis = simulator.transient(end_time=1e-9, step_time=1e-12)
  File "C:\Users\Andrew\Anaconda3\lib\site-packages\PySpice\Spice\Simulation.py", line 753, in transient
    return self._run('transient', *args, **kwargs)
  File "C:\Users\Andrew\Anaconda3\lib\site-packages\PySpice\Spice\NgSpice\Simulation.py", line 76, in _run
    raw_file = self._spice_server(spice_input=str(self))
  File "C:\Users\Andrew\Anaconda3\lib\site-packages\PySpice\Spice\NgSpice\Server.py", line 160, in __call__
    stderr)
NameError: The number of points was not found in the standard error buffer, ngspice returned:

[image]

### Steps to reproduce the behaviour

See test2.txt (rename to .py)
[test2.txt](https://github.com/FabriceSalvaire/PySpice/files/3575426/test2.txt)

This may well be about how I installed ngspic, but I couldn't find any good instructions on this.

First I installed the shared lib, by following the installation instructions here: https://pyspice.fabrice-salvaire.fr/installation.html

At this point the ngspice-shared simulator works, however ngspice-subprocess failed since it couldn't execute "ngspice". So I unzipped ngspice-30_64.zip to program files in the spice64 dir. I also had to add the bin dir to my PATH.

Now I get the above errors, and I'm kind of stuck.

Finally my reason for wanting to use the subprocess is I want to run a lot of simulations, and it's taking a while, so I figured I'd try multi-threading. However multithreading with a shared lib sounded like it might not work, and indeed it didn't, I got the following error (which I've seen other issues about):

  ...
  File "C:\Users\Andrew\Anaconda3\lib\site-packages\cffi\api.py", line 107, in cdef
    self._cdef(csource, override=override, packed=packed)
  File "C:\Users\Andrew\Anaconda3\lib\site-packages\cffi\api.py", line 121, in _cdef
    self._parser.parse(csource, override=override, **options)
  File "C:\Users\Andrew\Anaconda3\lib\site-packages\cffi\cparser.py", line 315, in parse
    self._internal_parse(csource)
  File "C:\Users\Andrew\Anaconda3\lib\site-packages\cffi\cparser.py", line 355, in _internal_parse
    decl.type, name=decl.name, partial_length_ok=True)
  File "C:\Users\Andrew\Anaconda3\lib\site-packages\cffi\cparser.py", line 587, in _get_type_and_quals
    tp = self._get_struct_union_enum_type('struct', type, name)
  File "C:\Users\Andrew\Anaconda3\lib\site-packages\cffi\cparser.py", line 734, in _get_struct_union_enum_type
    raise CDefError("duplicate declaration of struct %s" % name)
cffi.error.CDefError: <cdef source string>:7: duplicate declaration of struct ngcomplex

See test.txt (rename to .py) [test.txt](https://github.com/FabriceSalvaire/PySpice/files/3575470/test.txt)

I couldn't find any info on multi threading with PySpice, and so please tell me if there's a better way / docs on how to do this.

Thanks,
Andrew

rbarzic, 2020-02-28:
@andrewparlane , do you mind sharing a example of Pyspice + Multiprocessing ? I've been struggling with that for a while now and can't get it to work correctly

andrewparlane, 2020-03-02:
I'm afraid I don't have any of that code anymore. I gave up with this eventually. I can't remember if it was due to PySpice not doing what I wanted, or other design issues. I'm no expert in python so it could have been something unrelated.

Sorry I can't help.

FabriceSalvaire, 2020-05-03:
> NameError: The number of points was not found in the standard error buffer, ngspice returned:

Could be due to a newer Ngspice release

Multithreading is a complex topic to be documented