PySpice-org/PySpice #171: raise KeyError(name) KeyError: '1N4148' (libraries wont load)

alejandrobarredo, 2019-10-04:
### Environment (OS, Python version, PySpice version, simulator)
Windows 10
Python 3.6.7
Pyspice 1.3.2

### Expected Behaviour
Load the diode and run the analysis

### Actual Behaviour
Traceback (most recent call last):
  File "C:/Users/109459/Documents/TransistorNN/diodeTest.py", line 22, in <module>
    circuit.include(spice_library['1N4148'])
  File "C:\Users\109459\AppData\Local\Continuum\anaconda3\envs\SHAP\lib\site-packages\PySpice\Spice\Library.py", line 107, in __getitem__
    raise KeyError(name)
KeyError: '1N4148'

### Steps to reproduce the behaviour
import re
import numpy as np
import time
import warnings
import matplotlib.pyplot as plt
from PySpice.Doc.ExampleTools import find_libraries
from PySpice.Probe.Plot import plot
from PySpice.Spice.Library import SpiceLibrary
from PySpice.Spice.Netlist import Circuit
from PySpice.Physics.SemiConductor import ShockleyDiode
import sys
from PySpice.Unit import *
import absl.logging
logging.root.removeHandler(absl.logging._absl_handler)
absl.logging._warn_preinit_stderr = False

libraries_path = find_libraries()
spice_library = SpiceLibrary(libraries_path)

circuit = Circuit('Diode Characteristic Curve')

circuit.include(spice_library['1N4148'])

circuit.V('input', 'in', circuit.gnd, 10@u_V)
circuit.R(1, 'in', 'out', 1@u_Ω) # not required for simulation
circuit.X('D1', '1N4148', 'out', circuit.gnd)

analyses = {}

simulator = circuit.simulator(temperature=25, nominal_temperature=25)
analysis = simulator.dc(Vinput=slice(-2, 5, .01))

alejandrobarredo, 2019-10-04:
libraries_path gets the value 'C:\\Users\\109459\\Documents\\libraries'.
Even if I hardcode the folder with ngspice libraries like 
**libraries_path = os.path.join('C:/Spice64/bin-dll/ngspice.lib')**

Ricky-PV, 2020-04-22:
Same problem for me (Windows 10, python 3.6.9, pyspice 1.3.2) . I tried to follow the instructions in the FAQ on setting the NGspice library path but the following is returned

``import PySpice

PySpice.Spice.NgSpice.Shared.NgSpiceShared.LIBRARY_PATH
Traceback (most recent call last):

  File "<ipython-input-25-30c4a7c2aee8>", line 1, in <module>
    PySpice.Spice.NgSpice.Shared.NgSpiceShared.LIBRARY_PATH

AttributeError: module 'PySpice.Spice' has no attribute 'NgSpice'
``

Any idea what is wrong with my installation?

oberron, 2020-08-06:
Same problem for me (Google colab, python 3.6, pyspice 1.4.3).

following the tutorial `8.4.2. Diode Characteristic Curve` i get the same error message 

```
<ipython-input-3-c7852ea96bd4> in <module>()
      1 circuit = Circuit('Diode Characteristic Curve')
      2 
----> 3 circuit.include(spice_library['1N4148'])
      4 
      5 circuit.V('input', 'in', circuit.gnd, 10@u_V)

/usr/local/lib/python3.6/dist-packages/PySpice/Spice/Library.py in __getitem__(self, name)
    105             # print('Library {} not found in {}'.format(name, self._directory))
    106             # self._logger.warn('Library {} not found in {}'.format(name, self._directory))
--> 107             raise KeyError(name)
    108 
    109     ##############################################

KeyError: '1N4148'
```

any links to help resolve this?

chr1s1s1, 2020-10-22:
Hello,
I have exactly the same problem/setup as @oberron . Any tips or guidance gratefully received!

oberron, 2020-10-24:
@chr1s1s1 , I could resolve this by adding following in the .py

```
PySpice_examples_path = "./drive/My Drive/pySpice/examples"
libraries_path = join(PySpice_examples_path, 'libraries')
```

then creating manually under the PySpice_examples_path, following folder structure (with the 1N1418 being available at : https://github.com/FabriceSalvaire/PySpic/examples/libraries/diode/switching/1N4148.lib)

libraries
|----diodes
       |-----Switching
                |-------1N1418.lib

specdrake, 2020-11-05:
> @chr1s1s1 , I could resolve this by adding following in the .py
>
> ```
> PySpice_examples_path = "./drive/My Drive/pySpice/examples"
> libraries_path = join(PySpice_examples_path, 'libraries')
> ```
>
> then [...]

@oberron Sorry but the link isn't working(even after correcting ```PySpic``` to ```PySpice```)

PravinSabariBala, 2023-03-04:
> 

have resolved this issue , if yes could you tell how  you resolved?

sclaramunt, 2023-04-13:
I just ran into this problem but with the nmos transistor example. In my case, I am using anaconda. Using the code from the example, the path was configured at c:\users\user\anaconda3\lib\libraries, but the folder "libraries" did not existed and the execution of the code returned the error comented. The issue was solved by copying the library folder from the github repository to its corresponding place, as suggested by specdrake. Thanks!

GabrielFerreira22, 2024-06-20:
I have this same issue on google colab 
@sclaramunt i didn't understand how you solve it. Could you give a more detailed explanation?