PySpice-org/PySpice #172: behavioral source expression not passed correctly on linux shared

oe1rsa, 2019-10-21:
### Environment (OS, Python version, PySpice version, simulator)
ubuntu 19.04 python 3.7 PySpice 1.3.2 ngspice 30

### Expected Behaviour
[image]
### Actual Behaviour
[image]
### Steps to reproduce the behaviour
from pylab import *

import PySpice.Logging.Logging as Logging
logger = Logging.setup_logging()

from PySpice.Spice.Netlist import Circuit
from PySpice.Unit import *
from PySpice.Spice.BasicElement import BehavioralSource
    
circuit = Circuit("RC")

source = circuit.BehavioralSource('source', 'in', circuit.gnd,
                                   voltage_expression = "time^4*exp(-1000*time)")

circuit.R(1,       'in', 'out',        u_kOhm(9))
circuit.R(2,       'out', circuit.gnd, u_kOhm(1))

simulator = circuit.simulator(simulator="ngspice-shared")
transient = simulator.transient(step_time=u_ms(1e-3), end_time=u_ms(20))

clf()
grid(True)
plot(transient['out'].abscissa, transient['out'])

FabriceSalvaire, 2020-05-03:
Generated spice code

```
.title Pulse
Bsource in 0 v=time^4*exp(-1000*time)
R1 in out 9kOhm
R2 out 0 1kOhm
```

indead ngspice plot something different

FabriceSalvaire, 2020-05-04:
reported to upstream https://sourceforge.net/p/ngspice/bugs/490/ since investigation on PySpice didn't figure out what is wrong

FabriceSalvaire, 2020-05-05:
from ngspice
```
import locale
locale.setlocale(locale.LC_NUMERIC, 'C')
```

commit 641df12e (2020-05-05): fix #172 for LC_NUMERIC ngspice issue
closed 2020-05-05 by FabriceSalvaire