PySpice-org/PySpice #174: OSError: cannot load library '/usr/local/lib/libngspice.dylib'

NatarajPinakapani, 2019-10-25:
### Environment (Mac Catalina, Python version 3.7.4, PySpice version 1.3.2, ngspice version 31)

### Expected Behaviour

The following code should execute without any error

#!/usr/local/bin/ python3
####################################################################################################
import PySpice.Logging.Logging as Logging
logger = Logging.setup_logging()
from PySpice.Spice.Netlist import Circuit, SubCircuit, SubCircuitFactory
from PySpice.Unit import *
####################################################################################################

class VoltDiv(SubCircuitFactory):
    __name__='VoltDiv'
    __nodes__=('in1', 'out', 'gnd')
    def __init__(self, R1_r, R2_r):
        super().__init__()
        self.R(1, 'in1', 'out', R1_r)
        self.R(2, 'out', 'gnd', R2_r)
    ##End def __init__
##End Class VoltDiv

##Start Test Bench
circuit=Circuit('vd')
circuit.X('1', 'VoltDiv', 'in1', 'out', circuit.gnd )
circuit.V('input', 'in1', circuit.gnd,10)
circuit.subcircuit(VoltDiv(10@u_Ohm, 20@u_Ohm))
##End Test Bench

def Simulate(dut):
    ##Start Simulation 
    simulator = dut.simulator(temperature=25, nominal_temperature=25)
    ##End Simulation 

    ##Start Output Analysis
    analysis=simulator.operating_point()
    vout=round(float(analysis.out),2)
    for node in analysis.branches.values():
       itot =round(float(node)*-1,2)
    print(vout, itot)
    return(vout,itot)
Simulate(circuit)

### Actual Behaviour
OSError: cannot load library '/usr/local/lib/libngspice.dylib': dlopen(/usr/local/lib/libngspice.dylib, 2): Symbol not found: _omp_set_num_threads
  Referenced from: /usr/local/lib/libngspice.dylib
  Expected in: flat namespace
 in /usr/local/lib/libngspice.dylib

### Steps to reproduce the behaviour

python3 <filename> 
The above command reproduces the stated "Actual Behaviour"

closed 2019-11-05 by NatarajPinakapani

BadeSummer, 2020-04-16:
I meet the same problem as you. I can find directly the file "libngspice.dylib" is in /usr/local/lib. But as running the PySpice example still occur OSError like yours. Have you solved this problem?

andrecavalcante, 2024-06-25:
I have the same problem.