PySpice-org/PySpice #179: SinusoidalMixin format_spice_parameters incomplete AC statement

kemlath, 2020-01-05:
### Environment (OS, Python version, PySpice version, simulator)

macOS Mojave
python Python 3.7.6
Version: 1.3.2
ngspice version 31

### Problem description

Not sure if this is due to the use of ngspice version 31 compared the currently supported version 30 but the AC statement generated by PySpice does not work. 

The `filter/low-pass-rc-filter.py` sample fails due to an incomplete AC statement in the PySpice generated netlist:
```
.title Low-Pass RC Filter
Vinput in 0 DC 0V AC SIN(0V 1V 50Hz 0s 0Hz)
R1 in out 1kOhm
C1 out 0 1uF
.options TEMP = 25°C
.options TNOM = 25°C
.ic 
.ac dec 10 1Hz 1MegHz
.end
```

If I pass this netlist to ngspice directly it complains, but there is no error return from the shared library interface used by PySPice ?!?

The problem is in line: 
`Vinput in 0 DC 0V AC SIN(0V 1V 50Hz 0s 0Hz)`

the AC statement lacks the voltage amplitude i.e. to work, the line should read
`Vinput in 0 DC 0V AC 1V SIN(0V 1V 50Hz 0s 0Hz))`

The ngspice manual says it can be omitted and defaults to 1V but that does not seem to be the case.
 
### Solution

Change the format statement in SinusoidalMixin. format_spice_parameters as below by adding the Signal amplitude as AC amplitude setting

```
def format_spice_parameters(self):
        sin_part = join_list((self.offset, self.amplitude, self.frequency, self.delay, self.damping_factor))
        return join_list((
            'DC {}'.format(str_spice(self.dc_offset)),
            'AC {0} SIN({1})'.format(str_spice(self.amplitude), sin_part)
         #   'AC SIN({})'.format(sin_part),
        ))
```

With this modified format_spice_parameters function the `filter/low-pass-rc-filter.py`  sample works as expected.

FabriceSalvaire, 2020-05-03:
> f I pass this netlist to ngspice directly it complains, but there is no error return from the shared library interface used by PySPice ?!?

**Yes it is another issue ...**

FabriceSalvaire, 2020-05-03:
Do you understand the manual ???

FabriceSalvaire, 2020-05-03:
reported to https://sourceforge.net/p/ngspice/bugs/488/

FabriceSalvaire, 2020-05-03:
let see what they answer because it is hard to fix PySpice if the ngspice manual is voodoo

FabriceSalvaire, 2020-05-04:
fixed in 8d34ff6

commit 8d34ff6c (2020-05-04): fix #179
closed 2020-05-04 by FabriceSalvaire