PySpice-org/PySpice #181: Ngspice version 31 support

matthuszagh, 2020-01-24:
Would it be possible to add Ngspice version 31 support? Are there any specific hurdles to supporting it, or it just hasn't been tested yet?

### Steps to reproduce the behaviour
```
$ python
>>> from PySpice.Spice.NgSpice.Shared import NgSpiceShared
>>> ngspice = NgSpiceShared.new_instance()
Unsupported Ngspice version 31
```

stuarteberg, 2020-03-12:
Looks like that's [just a warning][1].  Aside from that warning, did you notice any other problems?  (I'm not a PySpice dev, I'm just curious.)

[1]: https://github.com/FabriceSalvaire/PySpice/blob/c0dc504b65bed3aad52d06d6dbe06844ade7a7ce/PySpice/Spice/NgSpice/Shared.py#L426

matthuszagh, 2020-03-12:
I didn’t find any specific issues, was mostly just curious if the dev had found issues or just hadn’t tested it yet.

stuarteberg, 2020-03-12:
Looks like it ought to work, as long as [these fields][1] haven't changed.  They need to match [this enum][2], and they do.

[1]: https://github.com/FabriceSalvaire/PySpice/blob/c0dc504b65bed3aad52d06d6dbe06844ade7a7ce/PySpice/Spice/NgSpice/SimulationType.py#L58-L82

[2]: https://github.com/imr/ngspice/blob/9238c0f9d44fb413c92311c0ac88d61d1cf5af88/src/include/ngspice/sim.h#L4-L28

I've opened PR #186 to eliminate the warning.

alexmyczko, 2020-03-20:
This thing outputs all 0s and that warning

```
import numpy as np
import matplotlib.pyplot as plt

from PySpice.Spice.Netlist import Circuit
from PySpice.Unit import *

circuit = Circuit('Low-Pass RC Filter')

circuit.SinusoidalVoltageSource('input', 'in', circuit.gnd, amplitude=1@u_V)
R1 = circuit.R(1, 'in', 'out', 1@u_kΩ)
C1 = circuit.C(1, 'out', circuit.gnd, 1@u_uF)

simulator = circuit.simulator(temperature=25, nominal_temperature=25)
analysis = simulator.ac(start_frequency=1@u_Hz, stop_frequency=1@u_MHz, number_of_points=1,  variation='dec')

print(np.absolute(np.array(analysis.out)))
```

instead I'm expecting something like this:

Node name 'in' is a Python keyword
[9.9998027e-01 9.9803191e-01 8.4673303e-01 1.5717673e-01 1.5913479e-02
 1.5915474e-03 1.5915494e-04]

stuarteberg, 2020-03-20:
Do you know if the behavior is different for ngspice 30?

alexmyczko, 2020-03-20:
Dunno, I'm on Ubuntu 18.04 and backported the sid https://packages.debian.org/search?keywords=ngspice 31.3, can try the buster 30.2...

stuarteberg, 2020-03-20:
Huh.  I can also reproduce the difference in behavior using your example.  
Unfortunately, I don't have the knowledge or resources to investigate this.

With ngspice 30:

```
$ ngspice --version
ngspice compiled from ngspice revision 30
Written originally by Berkeley University
Currently maintained by the NGSpice Project

Copyright (C) 1985-1996,  The Regents of the University of California
Copyright (C) 1999-2011,  The NGSpice Project

$ pip list | grep PySpice
PySpice         1.3.2

$ python /tmp/example.py
Node name 'in' is a Python keyword
[9.9998027e-01 9.9803191e-01 8.4673303e-01 1.5717673e-01 1.5913479e-02
 1.5915474e-03 1.5915494e-04]
```

But with ngspice 31:

```
$ ngspice --version
ngspice compiled from ngspice revision 31
Written originally by Berkeley University
Currently maintained by the NGSpice Project

Copyright (C) 1985-1996,  The Regents of the University of California
Copyright (C) 1999-2011,  The NGSpice Project

$ pip list | grep PySpice
PySpice         1.3.2

$ python /tmp/example.py
Node name 'in' is a Python keyword
Unsupported Ngspice version 31
[0. 0. 0. 0. 0. 0. 0.]
```

brainstorm, 2020-04-06:
@alexmyczko I've recently jumped to ngspice 31 and while testing I found this issue and playing with your example I  identified a couple of issues with your program, here's a slightly revised version, easier to debug:

```
#!/usr/bin/env python

import numpy as np
import matplotlib.pyplot as plt

from PySpice.Spice.Netlist import Circuit
from PySpice.Unit import u_V, u_kOhm, u_uF, u_Hz, u_MHz

circuit = Circuit('Low-Pass RC Filter')

circuit.SinusoidalVoltageSource('input', 'inp', circuit.gnd, amplitude=1@u_V)
R1 = circuit.R(1, 'inp', 'out', 1@u_kOhm)
C1 = circuit.C(1, 'out', circuit.gnd, 1@u_uF)

simulator = circuit.simulator(temperature=25, nominal_temperature=25)
analysis = simulator.ac(start_frequency=1@u_Hz, stop_frequency=1@u_MHz, number_of_points=1,  variation='dec')

print(analysis._simulation._ngspice_shared._stdout)
#print(np.absolute(np.array(analysis.out)))
```

But in essence, this is what happens under the hood on ngspice side:

```
./test_ngspice30.py 
['Circuit: Low-Pass RC Filter', 'Error on line 2 :', 'vinput inp 0 dc 0v ac sin(0v 1v 50hz 0s 0hz)', 'parameter value out of range or the wrong type', 'Doing analysis at TEMP = 25.000000 and TNOM = 25.000000', 'Reference value :  1.00000e+00\r', 'No. of Data Rows : 7']
```

Hope that helps! ;)

alexmyczko, 2020-04-06:
Frankly it wasn't *my* example, but a persons that I support, I sent them a link, however I'm not sure they will read it... But thanks!

Lewous, 2020-04-30:
The version warning seems comes from [here](https://github.com/FabriceSalvaire/PySpice/blob/master/PySpice/Spice/NgSpice/SimulationType.py), and have been fixed (but not on pypi).
I tried reinstall ngspice-30.pkg over version 31 on mac OS, but failed. Some examples still shows output is zero ( I got figures plotted  but only straight lines on there).

FabriceSalvaire, 2020-05-03:
fixed on a6476cf1f1ab47a5dc28da8d9d3a9eb72f5eee3e but not SIN folks

closed 2020-05-04 by FabriceSalvaire