PySpice-org/PySpice #183: Can't run examples

sakibmohammad, 2020-02-05:
Hi, I am new to Python. I am using PySpice for a project (I'm on Win10). I have installed every package correctly (that I know of). I am trying to run this example: https://github.com/FabriceSalvaire/PySpice/blob/master/examples/diode/voltage-multiplier.py and getting this error
[image]

What can I do? Any help will be really appreciated. 
Thanks.

stuarteberg, 2020-03-12:
I am not a spice expert, but I think the examples won't work unless you have a library of SPICE component models installed on your machine.  In the example you mentioned, the following lines are responsible for locating that library:

```
libraries_path = find_libraries()
spice_library = SpiceLibrary(libraries_path)
```

But some advice on where to find and install such models would be appreciated.

sakibmohammad, 2020-04-01:
Thank you for your response. I was able to run the examples. I was to change the source code of the library to make sure it can detect the ngspice library. Then I had to download the github repository of PySpice and,  had to change the following line, and add the library path manually. 

libraries_path = find_libraries()

closed 2020-04-01 by sakibmohammad