PySpice-org/PySpice #188: Simplify installation instructions with conda-forge

stuarteberg, 2020-03-12:
Good news!  There is now a [conda-forge package for `ngspice`][1] (mac/linux only at the moment).  That means the PySpice [installation instructions][2] for mac/linux can be simplified somewhat.  (For example, homebrew is no longer required on Mac.)

The following installation commands will work on Mac and Linux:
```
conda create -n spice -c conda-forge ngspice pyyaml cffi matplotlib numpy ply scipy
conda activate spice
pip install PySpice
```

Furthermore, once the PySpice release tags are updated (#187), then creating a conda package for `pyspice` itself will be trivial, and the installation instructions will become even simpler:

```
conda create -n spice -c conda-forge pyspice
conda activate spice
```

I'm happy to create a `pyspice` package for conda-forge, but I can't do so until #187 is resolved.

[1]: https://github.com/conda-forge/ngspice-feedstock/blob/master/recipe/meta.yaml
[2]: https://pyspice.fabrice-salvaire.fr/installation.html#installation-page

FabriceSalvaire, 2020-05-03:
I don't use personally conda on Linux because it is intended for scientific user and not developer

But would be nice to have a simple way to install PySpice on all platform especially on OSX and Windows

I will release a 1.3 soon

FabriceSalvaire, 2020-05-03:
on which platforms can you test the conda package ?

stuarteberg, 2020-05-03:
Conda-forge can support Mac, Linux, and windows, but for now the ngspice package in conda-forge only supports Mac and Linux. 

My recommendation is to start with just those two and think about Windows afterwards.

endolith, 2020-05-06:
Confirmed it doesn't work on Windows :(

> PackagesNotFoundError: The following packages are not available from current channels:
>
>  - ngspice

Anything I can do to help with that?

stuarteberg, 2020-05-06:
@endolith 

>Confirmed it doesn't work on Windows :(
>Anything I can do to help with that?

Yes! The [`ngspice` "feedstock" repo][1] contains the conda-build recipe for Mac and Linux.  It needs to be enhanced to support Windows, too.

Here are the steps to follow.  You can try looking at other feedstock repos as examples, such as the [`sqlite` feedstock][3].

1. Fork the feedstock repo.
2. Add a file named `recipe/bld.bat` which should build `ngspice` using visual studio, and install it to the appropriate directory.  On Mac/Linux, that directory is specified by the `$PREFIX` environment variable, but on Windows I think it's `LIBRARY_BIN`.  The environment variables available to `bld.bat` can be found on [this page of the conda-build docs][2].
3. Edit `recipe/meta.yaml`:
    - Delete the `skip: true` line entirely.
    - For any other lines that should not apply to the windows build, add a "selector" comment of the form: `# [not win]`
4. Submit a Pull Request to the main ngspice feedstock repo.  Examine the CI build logs and push changes until it passes.  I'll offer assistance there if I can find time.

I know `ngspice` supports a lot of optional features, some of which can complicate the build.  Let's aim for the bare minimum feature set to start with.  That may reduce the number of dependencies required by the build.

If you're already familiar with`conda-build`, you can try building the recipe locally before pushing your PR.  That might allow you to iterate slightly faster, but honestly it's probably easiest just to let the CI system run.

[1]: https://github.com/conda-forge/ngspice-feedstock
[2]: https://docs.conda.io/projects/conda-build/en/latest/user-guide/environment-variables.html#environment-variables-set-during-the-build-process
[3]: https://github.com/conda-forge/sqlite-feedstock

stuarteberg, 2020-05-15:
Update: PySpice conda packages are now published to conda-forge, so these commands now work on Mac and Linux:

```
conda create -n spice -c conda-forge pyspice
conda activate spice
```

The list of supported platforms (osx and linux), python versions (3.6, 3.7, 3.8), and PySpice versions (1.3.2, 1.4.2) can be inferred from the hosted package files:
https://anaconda.org/conda-forge/pyspice/files

Now the installation instructions can updated in the documentation (for Mac/Linux).

If someone wants to put some effort into Windows support, the first step is to enhance the `ngspice` recipe as described in the comment above.

FabriceSalvaire, 2020-05-15:
Some updates

- for Windows / OSX I would prefer to rely on upstream binaries, because they could maintain a binary for theses "normal" OS
- for Linux it should be the job of the distribution, and it seems better now
- **Something is wrong with the conda ngspice package, should not depend on X11etc.**

- I  started to implement a tool to download ngspice DLL from sourceforge and do the job in PySpice so as to avoid user mistakes ...
- Actually DLL is missing for 32 and I don't know what contains the OSX pack, but 32 is very recent ...

- I installed miniconda for test, added a meta.yaml in PySpice, pushed a conda/PyPI package after several try/fixes in PySpice, there is still some issues to fix ...

**I don't use Anaconda what is the difference between Ancaconda cloud and conda-forge ? I guess conda-forge automatically build the package from git update ?**

FabriceSalvaire, 2020-05-15:
I also started to rewrite the doc ...

FabriceSalvaire, 2020-05-15:
https://conda-forge.org
https://conda-forge.org/feedstocks
https://github.com/conda-forge/pyspice-feedstock

FabriceSalvaire, 2020-05-15:
Just one important point that I have missed ...

We should be able to install ngspice/PySpice without an admin account on a computer, especially a university one ! And without to compile something ! This is a good reason to have a conda package !

stuarteberg, 2020-05-16:
I will try to answer all of your questions but I think it's easiest start with the conda-forge questions first:

>what is the difference between Anaconda cloud and conda-forge ?

Yeah, sorry, this is often confusing for new conda users.  Let me define some terms.  You probably already know some of this, but I just want to make sure we're on the same page:

`conda` is a package manager that is completely independent of the OS package manager.  All packages it downloads are contained within a common base directory, which can be placed anywhere on your hard drive, without root permissions.  If necessary, you could delete them all with a single call to `rm -rf`.

By default, `conda` fetches packages from the main package repository (also called the `defaults` channel). 

By default, `conda` fetches packages from the main package repository (also called the `defaults` channel), but that's not the only place to find packages.  Anyone can create a personal account on `anaconda.org` ("Anaconda Cloud"), and upload their own packages to share.  Your personal account is called your "channel".  For instance, I have a channel named `flyem-forge`, and I previously uploaded an `ngspice` package there.  To install my particular build of `ngspice`, one could type:

```
conda install --channel flyem-forge ngspice

# or the shorter form:
conda install -c flyem-forge ngspice
```

The `defaults` channel contains most packages people want, and the versions (and build configurations) are carefully curated so that they will work well together.  In other words, it's a distribution or conda packages made by a company, i.e. Anaconda, Inc. (Somewhat analogous to the way RHEL is a distribution of RPM packages made by a company, i.e. RedHat.).

But there is also a large community project to create a distribution of conda packages, completely in the open.  That project is named conda-forge.  (Somewhat analogous to CentOS, which is a community-managed distribution of RPM packages.). On anaconda.org, the conda-forge channel has more packages than any other, even the defaults.  The conda-forge project uses a system of GitHub repositories ("feedstocks") and free CI services to maintain and build their distribution.

When you want to install conda for the first time, you have a choice of what to download:
- [Miniconda][], which provides the `conda` executable and a bare-minimum set of its dependencies, with nothing else pre-installed
- [The full Anaconda Installer][anaconda], which is basically what you would get if you downloaded Miniconda and then ran `conda install ...` with a very long list of packages from the `defaults` channel.  These packages are all pre-installed into the `base` conda environment.

[Miniconda]: https://docs.conda.io/en/latest/miniconda.html
[anaconda]: https://www.anaconda.com/products/individual#Downloads

It's usually best to install Miniconda, and then create an environment for your needs.  And I recommend using `conda-forge` instead of the `defaults` channel, if possible:

```
conda create -n my-spice-env pyspice ipython jupyterlab
conda activate my-spice-env
```

>I guess conda-forge automatically build the package from git update ?

Yes, as a convenience, there is a conda-forge bot that detects new release tags and creates new PRs based on the new release tags.  If the CI passes, then clicking "merge" adds the new version to the conda-forge channel.

>Can we push test package with conda-forge ?

What do you mean by "test package"?  The package is already tested as part of the build process, if that's what you mean.  The `pyspice` recipe [runs the following tests](https://github.com/conda-forge/pyspice-feedstock/blob/master/recipe/meta.yaml#L40-L58):

- Verify that all `PySpice` modules can be imported
- Run all tests in the PySpice `unit-test` directory (well, anything that `pytest` can find)

But if you're asking how to create a completely experimental version of the package for test purposes, then it's best to build the package locally on your own machine (by running `conda build ....` with the recipe on your own machine) and then upload that package to your personal channel (not the conda-forge channel).

stuarteberg, 2020-05-16:
>for Windows / OSX I would prefer to rely on upstream binaries, because they could maintain a binary for theses "normal" OS

>for Linux it should be the job of the distribution, and it seems better now

To ensure that conda-forge packages will run on as many other machines as possible, conda-forge requires that its packages depend on (almost) [no external libraries][external].

[external]: https://conda-forge.org/docs/maintainer/adding_pkgs.html#no-external-deps

This means the conda-forge `pyspice` package must depend on conda-forge's `ngspice` package, not some third-party build.

>I started to implement a tool to download ngspice DLL from sourceforge and do the job in PySpice so as to avoid user mistakes ...

As mentioned above, I think downloading a binary from source forge is not allowed for the conda-forge channel.  BUT if that's the best option at the moment, you CAN do that in your own recipe and upload it to your own channel on anaconda.org.  You won't benefit from the infrastructure that conda-forge provides, but at least you'll be able to give your users with a conda-based solution.

>Something is wrong with the conda ngspice package, should not depend on X11etc.

When I created that package, I just followed the directions in ngspice/INSTALL.  I guess the issue is that the `ngspice` executable requires X11, but `libngspice.so` does not?  We can discuss this in the [ngspice feedstock repo](https://github.com/conda-forge/ngspice-feedstock/issues/1).

>I installed miniconda for test, added a meta.yaml in PySpice, pushed a package on conda and PyPI after several try/fixes in PySpice, there is still some issues to be fixed ...

What are you trying to test?  The Linux package?  Or are you trying to add Windows support to the pyspice feedstock?  If the latter, try a pull request and we can examine the build failures there.

stuarteberg, 2020-05-27:
@FabriceSalvaire @endolith

Updates:

- Instead of a monolithic `ngspice` package that includes both `libngspice.so` and the `ngspice` executable, I've split the recipe into different packages: `ngspice-lib`, `ngspice-exe`, and (if desired) a combined `ngspice` package that is the superset of the other two.  Users who only need `libngspice.so` (e.g. PySpice users) can download `ngspice-lib`, which does not depend on X11.

- The conda-forge `ngspice` and `pyspice` packages now support Windows.  Windows users can install ngspice and PySpice in exactly the same way as Mac and Linux users:

    ```
    conda create -n spice -c conda-forge pyspice
    conda activate spice
    ```

    (For now, the only supported versions are `ngspice-32` and `PySpice-1.4.2`.)

    ### NOTE: I haven't tested the Windows package myself

    ...but it does pass all of PySpice's unit-tests and also passes PySpice's check-install script.

FabriceSalvaire, 2020-07-14:
done

closed 2020-07-14 by FabriceSalvaire