PySpice-org/PySpice #189: "NameError: Netlist is empty" when using BSIM3 modelcards (and solution ?)

cyber-g, 2020-03-20:
### Environment (OS, Python version, PySpice version, simulator)
Debian GNU/Linux 10 (buster)
Python 3.7.3
PySpice version : PySpice master on 20 March 2020 (version 1.3.2)

### Problem statement and Actual Behaviour
I would like to use BSIM3 modelcards found in the archive : 
 http://bsim.berkeley.edu/BSIM4/BSIM3/ftpv330.zip
as "ftpv330/test/modelcard.nmos" and "ftpv330/test/modelcard.pmos"

When I put those files in the directory : 
PySpice/examples/libraries/transistor/
and rename as "*.mod" files, the example 
PySpice/examples/transistor/transistor.py 
fails to run : 
```
python3 PySpice/examples/transistor/transistor.py
2020-03-20 - PySpice.Doc.ExampleTools.find_libraries - INFO - SPICE library path is /root/PySpice/examples/libraries
Traceback (most recent call last):
  File "PySpice/examples/transistor/transistor.py", line 30, in <module>
    spice_library = SpiceLibrary(libraries_path)
  File "/root/PySpice/PySpice/Spice/Library.py", line 76, in __init__
    spice_parser = SpiceParser(path)
  File "/root/PySpice/PySpice/Spice/Parser.py", line 805, in __init__
    self._statements = self._parse(lines)
  File "/root/PySpice/PySpice/Spice/Parser.py", line 846, in _parse
    raise NameError('Netlist is empty')
NameError: Netlist is empty
```

### Expected Behaviour
The example should execute properly ; I just added some files in the library directory. 

### Steps to reproduce the behaviour
I have prepared a Docker file to reproduce the behaviour. 
[Dockerfile.txt](https://github.com/FabriceSalvaire/PySpice/files/4360589/Dockerfile.txt)

### Solution
The solution to the problem is given as comments in the end of the dockerfile.

In the Parser class, there is a test on the length of the netlist after formatting '+'-lines and removing comments. 
In my case, the netlist was only one single line. And it triggers the error `NameError('Netlist is empty')` . 

That is why I propose to fix this issue by replacing the line 
`if len(lines) <= 1:`
by : 
`if len(lines) < 1:`

Best regards

tvrusso, 2020-03-20:
What I'm about to write may not be the problem you're describing, but it sounds like it might be.

Standard SPICE netlist parsing considers the very first line of a netlist file as a comment (the title of the run).  This has been true for as long as SPICE has existed.  So if your top-level netlist file only has one line, the parser is correct to be ignoring that one line.

If that's the issue, the correct solution is to add a title line to your netlist, not to modify the parser.

cyber-g, 2020-03-21:
> when using BSIM3 modelcards (and solution ?) (#189)

Dear Tom, 
Thanks for your reply ; your remark matches exactly the situation I have. 

Looking again at the file example, the netlist has no title and begins directly with '.model ...' 

I did not know the standard 1st line comment. 

I will modify the netlist so that it is compliant. 

It may be interesting to specify this element in the documentation. 

How can I properly contribute ? In which file of the repository should I put this documentation element ? 

(Sorry, I am beginning with this project and also beginner with python) 

Best regards 
Germain 

----- Mail original -----

cyber-g, 2020-04-22:
Looking more closely to the PySpice library code (SpiceParser._parse()), I have seen that models files must have a comment at the first line
Best regards

closed 2020-04-22 by cyber-g