PySpice-org/PySpice #193: ModuleNotFoundError: No module named 'pyspice'

hillmanis, 2020-04-11:
### Environment: 
(Windows 10, Python 3.7.6, PySpice 1.3.2, Ngspice)

### Description of issue:
I have just installed Pyspice via pip following the instructions at: https://pyspice.fabrice-salvaire.fr/installation.html

Using Python in a Jupyter notebook I attempt to load pyspice by typing: "import pyspice as ps" and running. This results in the error:

###Error Message:
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-2-0eab7ae5ac5d> in <module>
----> 1 import pyspice as ps

ModuleNotFoundError: No module named 'pyspice'

**Attempts to remedy the issue:**
- I have checked in the Anaconda navigator to verfiy that the  package is installed.
- Reinstalled pyspice (via pip).
- Uninstalled and re-installed Anaconda followed by re-installation of pyspice.

**pip installation output:**
(base) C:\Users\cehillma>pip install PySpice
Requirement already satisfied: PySpice in c:\users\cehillma\anaconda3\lib\site-packages (1.3.2)
Requirement already satisfied: numpy in c:\users\cehillma\anaconda3\lib\site-packages (from PySpice) (1.18.1)
Requirement already satisfied: PyYAML in c:\users\cehillma\anaconda3\lib\site-packages (from PySpice) (5.3)
Requirement already satisfied: scipy in c:\users\cehillma\anaconda3\lib\site-packages (from PySpice) (1.4.1)
Requirement already satisfied: matplotlib in c:\users\cehillma\anaconda3\lib\site-packages (from PySpice) (3.1.3)
Requirement already satisfied: cffi in c:\users\cehillma\anaconda3\lib\site-packages (from PySpice) (1.14.0)
Requirement already satisfied: ply in c:\users\cehillma\anaconda3\lib\site-packages (from PySpice) (3.11)
Requirement already satisfied: python-dateutil>=2.1 in c:\users\cehillma\anaconda3\lib\site-packages (from matplotlib->PySpice) (2.8.1)
Requirement already satisfied: cycler>=0.10 in c:\users\cehillma\anaconda3\lib\site-packages (from matplotlib->PySpice) (0.10.0)
Requirement already satisfied: kiwisolver>=1.0.1 in c:\users\cehillma\anaconda3\lib\site-packages (from matplotlib->PySpice) (1.1.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in c:\users\cehillma\anaconda3\lib\site-packages (from matplotlib->PySpice) (2.4.6)
Requirement already satisfied: pycparser in c:\users\cehillma\anaconda3\lib\site-packages (from cffi->PySpice) (2.19)
Requirement already satisfied: six>=1.5 in c:\users\cehillma\anaconda3\lib\site-packages (from python-dateutil>=2.1->matplotlib->PySpice) (1.14.0)
Requirement already satisfied: setuptools in c:\users\cehillma\anaconda3\lib\site-packages (from kiwisolver>=1.0.1->matplotlib->PySpice) (45.2.0.post20200210)

cyber-g, 2020-04-17:
Maybe your PYTHONPATH environment variable is not properly set. 

Please, can you execute the following commands in your Python shell : 
```python
import sys
print '\n'.join(sys.path)
```

and paste the output ?

FabriceSalvaire, 2020-05-03:
module is **PySpice**

hillmanis, 2020-05-09:
Thank you for your email. The following is the output of sys.path:

C:\\Users\\username\\Anaconda3\\python37.zip

C:\\Users\\username\\Anaconda3\\DLLs

C:\\Users\\username\\Anaconda3\\lib

C:\\Users\\username\\Anaconda3

C:\\Users\\username\\Anaconda3\\lib\\site-packages

C:\\Users\\username\\Anaconda3\\lib\\site-packages\\win32

C:\\Users\\username\\Anaconda3\\lib\\site-packages\\win32\\lib

C:\\Users\\username\\Anaconda3\\lib\\site-packages\\Pythonwin

cyber-g, 2020-05-11:
Dear @hillmanis, 
as pointed out by FabriceSalvaire, in order to use PySpice, you should import the module with its exact name (be careful of capitals, it is case sensitive). 
And it is not a problem of PYTHONPATH since the output of `'\n'.join(sys.path)` is OK. 
Please try the following command in your python shell : 
``` python
import PySpice as ps
```
or
``` python
from PySpice.Spice.Netlist import Circuit
```

FabriceSalvaire, 2020-05-15:
Discovered yesterday that Anaconda lowercase package name, could be error prone ...

closed 2020-05-15 by FabriceSalvaire