PySpice-org/PySpice #199: Problem with sinusoidal wave

rodrigo103, 2020-05-02:
### Environment (OS, Python version, PySpice version, simulator)
Kubuntu 18.04 LTS, Python version: 3.8.2 64 bit, PySpice: 1.3.2, ngspice: 31

### Expected Behaviour
I expected this example to run OK:
https://github.com/FabriceSalvaire/PySpice/blob/master/examples/filter/rlc-filter.py

### Actual Behaviour
It gave me the following error:

```
Error on line 2 :
    vinput in 0 dc 0v ac sin(0v 1v 50hz 0s 0hz)
```

So i tracked it down and figured out that ngspice was throwing the error.
Apparently there is a problem with how it is calling the sin function.

### Steps to reproduce the behaviour

Try to run the example with the configuration that i provided before.

According to NGSPICE 30 user manual the voltage sources have this general form:

```
VXXXXXXX N+ N- <<DC> DC/TRAN VALUE> <AC <ACMAG <ACPHASE>>>
+ <DISTOF1 <F1MAG <F1PHASE>>> <DISTOF2 <F2MAG <F2PHASE>>>
```

It says:

> ACMAG is the ac magnitude and ACPHASE is the ac phase. The source is set to this value in the ac
> analysis. If ACMAG is omitted following the keyword AC, a value of unity is assumed. If ACPHASE
> is omitted, a value of zero is assumed. If the source is not an ac small-signal input, the keyword
> AC and the ac values are omitted.

Being ACMAG and ACPHASE optionals, what pyspice generated seems ok, but it is not ok for ngspice, so i dont know which one, either pyspice or ngspice is wrong.

I solved it temporarily by editing https://github.com/FabriceSalvaire/PySpice/blob/master/PySpice/Spice/HighLevelElement.py

changed 
``` python
    def format_spice_parameters(self):

        sin_part = join_list((self.offset, self.amplitude, self.frequency, self.delay, self.damping_factor))
        return join_list((
            'DC {}'.format(str_spice(self.dc_offset)),
            'AC SIN({})'.format(sin_part),
        ))
```
by 
``` python
    def format_spice_parameters(self):

        sin_part = join_list((self.offset, self.amplitude, self.frequency, self.delay, self.damping_factor))
        return join_list((
            'DC {}'.format(str_spice(self.dc_offset)),
            'AC 1 SIN({})'.format(sin_part),
        ))
```

I created a pull request with this change: https://github.com/FabriceSalvaire/PySpice/pull/198

FabriceSalvaire, 2020-05-03:
Plot are wrong but I don't get an error ??? Must investigate ...

FabriceSalvaire, 2020-05-03:
from v31 manual

General form:
VXXXXXXX N+ N- <<DC> DC/TRAN VALUE> <AC <ACMAG <ACPHASE>>>
+ <DISTOF1 <F1MAG <F1PHASE>>> <DISTOF2 <F2MAG <F2PHASE>>>
IYYYYYYY N+ N- <<DC> DC/TRAN VALUE> <AC <ACMAG <ACPHASE>>>
+ <DISTOF1 <F1MAG <F1PHASE>>> <DISTOF2 <F2MAG <F2PHASE>>>

Examples:
VIN 13 2 0.001 AC 1 SIN(0 1 1MEG)

ACMAG is the ac magnitude and ACPHASE is the ac phase. The source is set to this value in the ac
analysis. If ACMAG is omitted following the keyword AC, a value of unity is assumed. If ACPHASE
is omitted, a value of zero is assumed. If the source is not an ac small-signal input, the keyword
AC and the ac values are omitted.

rodrigo103, 2020-05-03:
Another user reported the same problem before with the same PySpice and ngspice versions as me:
https://github.com/FabriceSalvaire/PySpice/issues/179

With which version of ngspice did you run the example?

FabriceSalvaire, 2020-05-03:
v31

FabriceSalvaire, 2020-05-03:
Do you understand the manual ???

FabriceSalvaire, 2020-05-03:
reported to https://sourceforge.net/p/ngspice/bugs/488/

FabriceSalvaire, 2020-05-03:
duplicate of #179

closed 2020-05-03 by FabriceSalvaire

rodrigo103, 2020-05-03:
Not fully. 
Correct me if i am wrong, but it seems that to specify DC/TRAN VALUE one must put the DC keyword before. The same for ACMAG, it seems to need the AC keyword, and ACPHASE seems to need a ACMAG specified. But this is just a guess.
Thank you for reporting it to ngspice. Maybe they can point out what is wrong.